bambulab/BambuStudio #2477: Bambu studio prevents windows monitors from turning off if video preview is running

mbilsky, 2023-09-07:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

Print a part. Start the live video preview. 

It doesn't matter if printer is printing or not, nor if bambu studio is the active window or minimized.

If you walk away from your computer, the system won't turn off the monitors (like it normally would)

After a few weeks of this and not realizing what was going on, ran the command here: https://superuser.com/a/1711887

Once it showed Bambu Studio (output placed in log upload section) I tried pausing the video and the system went back to turning the monitors off as expected.

### Actual results

Screen doesn't turn off / go to sleep

### Expected results

Monitors should turn off due to inactivity

### Project file & Debug log uploads

C:\Windows\system32>powercfg /requests
DISPLAY:
[PROCESS] \Device\HarddiskVolume3\Program Files\Bambu Studio\bambu-studio.exe

SYSTEM:
[PROCESS] \Device\HarddiskVolume3\Program Files\Bambu Studio\bambu-studio.exe

AWAYMODE:
None.

EXECUTION:
None.

PERFBOOST:
None.

ACTIVELOCKSCREEN:
None.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-09-08:
Same as #2011

GuzziRaz, 2023-09-09:
This is a user preference - I personally want it as is. You could make it optional though.

kylek29, 2023-09-12:
As a temporary workaround you can override this behavior using an administrator cmd prompt:

`powercfg /requestsoverride PROCESS "bambu-studio.exe" DISPLAY` 

That'll stick in perpetuity.

mbilsky, 2023-09-14:
I appreciate all the suggestions and feedback.

I understand how it is a feature that when Bambu Slicer is in the foreground and playing the video (e.g. not minimized) that you want it to keep the computer from turning off the video feed. Like a CCTV feed. And I have multiple monitors so also agree that if it is up on another monitor playing, but you're working or using a different one, that it keeping the monitors from sleeping could be a useful default.

But if it is minimized or other windows are on top of it, so you can't even see the video, then, at least for me, I'd want the screen to turn off.

I regularly hit (win+d) to minimize all windows when walking away from my computer to theoretically ensure that no streaming, youtube, browser tab, etc. is in the foreground that would prevent monitors from turning off.

TL:DR-- I think the default should be that if it is minimized, the screen should turn off

But having it as an option is a useful solution as well

AdrianGarside, 2023-10-07:
The bug is much worse - it isn't just stopping the display turning off/machine sleeping while actively watching the view which is arguably a correct behavior, but frequently (usually even) it leaks that state and permantently blocks the display/computer from turning off.

bambu123, 2023-10-08:
I understand. This needs to be checked again.

PunaJussi, 2024-06-06:
Even if I **don't** run the video, the monitor/system won't go to sleep.

SpikedSynapse, 2024-06-16:
Still an issue.

jpearce73, 2024-07-11:
I want to up vote this to be fixed. 
Its categorized as a new feature, I consider disabling the sleep function incorrectly is more of a bug and ask for attention, please.

bryanhunwardsen, 2024-08-02:
There is still a bug here(!)

I have not isolated it exactly but have given repro variations below, it is either to do with a disconnected device, and/or multiple instance of BS running, and/or multiple instance of BS running across multiple monitors.

Repro steps:
1) 2-3 instances of BambuStudio running in windows 10 with different projects open (all actively identifying the single printer I have) with different instance mixed in the foreground/background of multiple monitors on Windows 10.
2) Start video stream on one instance 
3) Possibly variations/combinations to trigger it:
 a) Stop the stream? 
 b) Stop and start on the different instance of BambuStudio
 c) Do start/stop actions across multiple instances of BS running on different monitors
 d) (!!!) Have one (or more) of just the actively selected printer get disconnected from BambuStudio (i.e. power loss, wifi issue, firmware lock-up on the printer requiring a power cycle)
 
Under the "d" option, since my only printer is showing disconnected, I cant attempt to start(then stop) the video stream of a disconnected printer and the the lock remains orphaned - forcing me to quit all instance of bambustudio to remove the power state screen on lock.