bambulab/BambuStudio #2479: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

iridesc, 2023-09-07:
### Bambu Studio Version

Bambu_Studio_linux_ubuntu-v01.07.04.52_538e9789b3d4e6c6dfd560cdb229231e.AppImage 

### Where is the application from?

Bambu Lab github releases

### OS version

DISTRIB_ID="ManjaroLinux" DISTRIB_RELEASE="23.0.0" DISTRIB_CODENAME="Uranos" DISTRIB_DESCRIPTION="Manjaro Linux"

### Additional system information

Linux ird-ms7a40 6.1.49-1-MANJARO #1 SMP PREEMPT_DYNAMIC Sun Aug 27 23:08:04 UTC 2023 x86_64 GNU/Linux

### Printer

p1p

### How to reproduce

1. download https://objects.githubusercontent.com/github-production-release-asset-2e65be/511797274/03a93af5-bf54-47f9-8de3-ad7e8f84a2c2?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230907%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230907T171357Z&X-Amz-Expires=300&X-Amz-Signature=99a43e9966e87f10541353bd107f0dfa1eb94d531805f645b0b1352f28505d86&X-Amz-SignedHeaders=host&actor_id=26148052&key_id=0&repo_id=511797274&response-content-disposition=attachment%3B%20filename%3DBambu_Studio_linux_ubuntu-v01.07.04.52.AppImage&response-content-type=application%2Foctet-stream
2.  chmod +x Bambu_Studio_linux_ubuntu-v01.07.04.52_538e9789b3d4e6c6dfd560cdb229231e.AppImage
3. ./Bambu_Studio_linux_ubuntu-v01.07.04.52_538e9789b3d4e6c6dfd560cdb229231e.AppImage

### Actual results

error result in current user：
[image]

with sudo have some error printing but working fine：
[image]

### Expected results

like with sudo

### Project file & Debug log uploads

[2023-09-08 01:14:58.133203] [0x00007fabedef7480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:31433): GLib-GObject-CRITICAL **: 01:14:58.280: invalid (NULL) pointer instance

(process:31433): GLib-GObject-CRITICAL **: 01:14:58.281: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

### Checklist of files to include

- [X] Log file
- [ ] Project file

iridesc, 2023-09-12:
update:  after ```rm -rf .config/BambuStudio``` it`s started. but when i login, the it's get stuck and exited.... the OrcaSlicer got the same problem

randerzander, 2023-09-20:
I have the same issue using Bambu_Studio_linux_ubuntu-v01.07.04.52.AppImage

viveksood97, 2023-09-21:
I am having the same issue

Achllle, 2023-11-04:
Same with [1.7.7.89](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.07.89). Removing `~/.config/BambuStudio` did not fix it for me, failing instead with a SegFault

```
(bambu-studio:164207): Gtk-CRITICAL **: 11:43:42.824: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
EGLDisplay Initialization failed: EGL_BAD_ACCESS
EGLDisplay Initialization failed: EGL_NOT_INITIALIZED
Cannot create EGL context: invalid display (last error: EGL_SUCCESS)
Segmentation fault (core dumped)
```

lanewei120, 2023-11-06:
which OS are you using @noisersup @Achllle @randerzander @viveksood97 ?

wad the homepage shown when crashed?
and if yes, did you click login?

thanks

Achllle, 2023-11-06:
I'm on Ubuntu 22.04. The app does not launch, only the following message is shown:
```
To manually specify the system certificate store, set the SSL_CERT_FILE environment variable to the correct CA 
```
Once you select 'yes', it crashes.

deftdawg, 2023-11-22:
@Achllle It's basically crashing as soon as the dialog closes and bambu-studio tries to start doing EGL/OpenGL graphical rendering

Can you run gdb and check if you're also seg-faulting on this function on [ImFont::CalcTextSizeA](https://github.com/bambulab/BambuStudio/blob/c5a79982860eb5486a2f90e812f0ec4e92a694db/src/imgui/imgui_draw.cpp#L3451) like [I am seeing on NixOS](https://github.com/NixOS/nixpkgs/pull/206495#issuecomment-1823227543):

```sh
# Open a gnome-terminal, konsole or xterm
sudo apt install -y gdb # Install GNU Debugger
echo -e "run\nbt\n\n" | LC_ALL=C gdb bambu-studio # Run bambu-studio inside GDB and dump the backtrace
```

deftdawg, 2023-11-28:
@Achllle the problem I described with EGL/OpenGL has been resolved in the NixOS's [bambu-studio PR](https://github.com/NixOS/nixpkgs/pull/206495/commits/a0ebeb6e1293f0bf2e37c9aae4b07a8b4e6e7aa4) based on some fixes upstream in prusa slicer to do with overriding the defaults for wgtk3.1...   No idea if that was your issue though, as you didn't reply. :laughing:

lanewei120, 2024-08-08:
Hello guys,

Do you still has this issue?

j4k0xb, 2024-08-12:
> Do you still has this issue?

I came across this issue today.
The root cause was an empty (0B) `.config/BambuStudio/BambuStudio.conf` file (was probably interrupted while saving).
After deleting the whole directory, everything (including login) worked again.

tylerburleigh, 2024-10-07:
I'm having this issue with the latest flatpak.

```
$ flatpak run com.bambulab.BambuStudio
$ [2024-10-07 19:12:51.285025] [0x00007198ccf6f3c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:4): GLib-GObject-CRITICAL **: 19:12:51.317: invalid (NULL) pointer instance

(process:4): GLib-GObject-CRITICAL **: 19:12:51.317: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

MackBambu, 2024-10-23:
You can try deleting the configuration file, as it might be failing to read due to an error in the file.
```rm -rf ~/.config/BambuStudio```

tylerburleigh, 2024-10-25:
I have tried deleting the config file, no luck.

dayer4b, 2024-11-05:
I believe this is affecting me too.  I have tried downgrading with no success so far.

Tnthr, 2024-11-24:
I had the same issue and as others noted the fail message is unrelated to this fault. The failure to launch might be caused by a corrupted conf file. My hard drive was full and it failed to write the config so both of my BambuStudio.conf and network plugin conf files were corrupted. I deleted those files and everything started working again.

If you are using a flatpak then you'll find the conf files in the below directory.

`~/.var/app/com.bambulab.BambuStudio/config/BambuStudio/`

MackBambu noted the normal install config file location above.

I would recommend better error handling of the config file being corrupted. Currently it seems to just fail silently.

matteovidali, 2025-02-24:
I was having this issue.
Steps I took to resolve
1. Delete the `~/.config/BambuStudio` directory via `rm -rf ~/.config/BambuStudio`
2. Delete the config file from `.var/app/com.bambulab.BambuStudio/config/BambuStudio/BambuStudio.conf` as mentioned by @Tnthr 

Bambu studio now launches while I have only standard user level privaleges as expected, with only the *usual* amount of gtk critical warnings. :)

Hope this was helpful!

will-x86, 2025-04-11:
> [@Achllle](https://github.com/Achllle) It's basically crashing as soon as the dialog closes and bambu-studio tries to start doing EGL/OpenGL graphical rendering
> 
> Can you run gdb and check if you're also seg-faulting on this function on [ImFont::CalcTextSizeA](https://github.com/bambulab/BambuStudio/blob/c5a79982860eb5486a2f90e812f0ec4e92a694db/src/imgui/imgui_draw.cpp#L3451) like [I am seeing on NixOS](https://github.com/NixOS/nixpkgs/pull/206495#issuecomment-1823227543):
> 
> # Open a gnome-terminal, konsole or xterm
> sudo apt install -y gdb # Install GNU Debugger
> echo -e "run\nbt\n\n" | LC_ALL=C gdb bambu-studio # Run bambu-studio inside GDB and dump the backtrace
> 
> **NOTE:** You might have to change **bambu-studio** to be the path to your AppImage or where-ever you're running it from

[bambu.log](https://github.com/user-attachments/files/19703046/bambu.log)
I've tried adding amdgpu to my kernel initd, I've tried updating to latest kernel, unstable and stable and deleted all of the cache each time 

I'm kinda stumped

will-x86, 2025-04-11:
Unfortunately for me I had both stable and unstable break for me 

AFAIK there are three options:
- Flatpak https://nixos.wiki/wiki/Flatpak
- Docker bambu studio image https://hub.docker.com/r/linuxserver/bambustudio
- Rebuild to older version

I chose to rebuild to an older version:
( This is my homeManager config ) 
```
custom-bambu-studio = pkgs.bambu-studio.overrideAttrs (oldAttrs: {
    version = "01.00.01.50";
    src = pkgs.fetchFromGitHub {
      owner = "bambulab";
      repo = "BambuStudio";
      rev = "v01.00.01.50";
      hash = "sha256-7mkrPl2CQSfc1lRjl1ilwxdYcK5iRU//QGKmdCicK30=";
    };
  });
in {
  imports = [
    base
  ];
///////////

  home.packages = with pkgs; [
    custom-bambu-studio
```
( here - https://github.com/will-x86/nixos-dotfiles/blob/81cd4018498e62fd1da5e0716a04e52785b79a55/home/desktop/desktop.nix#L15 ) 

This build created many issues for me as my machine had only 16GB of ram, so I had to use a server to remotely build it, see here:
https://github.com/will-x86/nixos-dotfiles/blob/81cd4018498e62fd1da5e0716a04e52785b79a55/rebuild.sh#L38C1-L38C142
and here:
https://github.com/will-x86/nixos-dotfiles/blob/81cd4018498e62fd1da5e0716a04e52785b79a55/hosts/framework/configuration.nix#L33

This fix kinda sucks, but is acceptable for now :)

Indomitable, 2025-05-26:
This issue happen often, but don't need to delete whole config folder you are going to loose everything.
If bambu studio stops opening and you check that `$HOME/.var/app/com.bambulab.BambuStudio/config/BambuStudio/BambuStudio.conf` has size of zero bytes, then delete only the `BambuStudio.conf` file and start again.