bambulab/BambuStudio #2480: Travel path makes strange artifacts on print.

kirek007, 2023-09-07:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P

### How to reproduce

1. Import model
2. Slice it
3. Eable travel path view
4. Look for strange circles 

### Actual results

Strange artifacts on print.

### Expected results

No artifacts on walls

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kirek007, 2023-09-07:
[image]
[image]

kirek007, 2023-09-07:
[image]

This is how it looks on ready print.
It's not bad here.. But I had huge problems on other models

QingZhangBambu, 2023-09-08:
are you use PLA Areo?

kirek007, 2023-09-08:
Yes, this is LW PLA.

QingZhangBambu, 2023-09-11:
this circle is the lift of layer change. And it seems that the parameters of the filament need to be adjusted.