bambulab/BambuStudio #2481: Crash when pressing Enter on height range modifier

edog1973, 2023-09-07:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Additional system information

n/a

### Printer

X1 Carbon

### How to reproduce

- Open a new project
- Insert a Bambu Cube object
- Add a Height Range Modifier
- Select the text box for the upper or lower range of the Height Range Modifier
- Type a number
- Press Enter 
- Application crashes, window closes with no error message,
[image]

WORKAROUND: Instead of pressing Enter click on any other UI element.  The updated value is accepted and the applicaiton does not crash

### Actual results

Application crashes

### Expected results

Application should not crash

### Project file & Debug log uploads

Project File: Not needed.
Debug Log: [debug_Wed_Sep_06_17_05_26_46952.log.zip](https://github.com/bambulab/BambuStudio/files/12553860/debug_Wed_Sep_06_17_05_26_46952.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

xp-Yang, 2023-11-06:
We fixed the issue. you can use the height range modifier normally in the next release.

LiZ-Li-BBL, 2024-01-16:
The issue has been fixed.

closed 2024-01-16 by LiZ-Li-BBL