bambulab/BambuStudio #2484: Arachne slicing odd behavior

kvnper, 2023-09-08:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

cpu: intel 6800k
gpu: 1080ti
RAM: 64gb

### Printer

X1C

### How to reproduce

Slice attached project file, observe slice model.

### Actual results

When using Arachne wall generator, outer walls are misplaced in a certain position on the model - they overlap the inner walls. It's as if they were placed on the wrong side. Problem goes away when using Classic wall generator.
[image]

### Expected results

Outer walls placed correctly.

### Project file & Debug log uploads

[log 230908 1317.zip](https://github.com/bambulab/BambuStudio/files/12555442/log.230908.1317.zip)
[230908 bambu studio arachne bug.zip](https://github.com/bambulab/BambuStudio/files/12555443/230908.bambu.studio.arachne.bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ktbcoach, 2023-10-04:
I've been able to recreate this issue when switching to the Arachne wall generator. it sporadically shows up on individual layers but all issues occur at the same corner of the part.

- The same stl files slices without issue in Cura and Prusa Slicer with Arachne settings so appears to be a Bambu fork issue.
- On layers where the issue occurs, the proceeding wall segment is performed backwards making the following G1 (which would actually correct if starting from the corner of the part) make a short cut across the part. 
- may not be significant, but on layers where this issue occurs there is no z seam for the wall indicated the GUI

[image]

QingZhangBambu, 2023-10-10:
there something wrong in clipper lib we used. it's a bit difficult to fix.

QingZhangBambu, 2023-10-10:
i will try to fix it.

kvnper, 2024-08-06:
Well we are awaiting a response

QingZhangBambu, 2024-08-23:
@kvnper sorry for the robot mark it inactive. it's error calculate result by clipper lib. i am try to replace the error function.

QingZhangBambu, 2024-08-23:
[image]
i will try to fix it as fast as possible

kvnper, 2024-08-23:
No rush on my end, I have not encountered the issue again. I just wanted to make sure it was not closed erroneously

QingZhangBambu, 2024-10-15:
it had been fixed, and will be release in the next version, please wait.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz