bambulab/BambuStudio #2485: Print Job Queuing

DillanS212, 2023-09-08:
I've been using a print queue plugin for Octoprint for my other printers for another year, and now that I'm running a pair of Carbon X1's alongside my other printers, I'm beginning to realise how much I rely on it for an efficient workflow. 

I'd like to see something like this integrated into the Bambu ecosystem. 

The workflow should work something like this: 

- When multiple plates are in the slicer, there is an option for 'Queue all plates.' 
- Printer allows you to map AMS as usual for each plate.
- Printer begins printing the first plate
- Printer completes print and notifies user via handy app
- User confirms plate is cleared
- Printer asks if the print was successful, if not, asks the user if they'd like to retry print.
- Printer prompts if filament needs to be changed, and tells the user what plate is required (textured, engineering etc.)
- Printer starts next (or retried) print
- Repeat until all queued jobs are completed. 

The user should also be able to: 

- Add prints to the queue 
- Re-order the queue
- Remove prints from the queue
- Change the quantity of each plate required (for printing multiples)

In a perfect world, the queue automatically sends prints to the next available printer omn the network with the correct filament loaded/nozzle fitted. 

This would make the Bambus the perfect print farm printer and would make volume part production much more practical. It's about all that's stopping me from swapping to Carbon X1's exclusively.

Piotrek912, 2023-12-30:
I need this function, currently it just impossible to use bambulab printers on printfarms... I mean 30+ printers, and 1000s of different projects. Ammount of job is just too big.

colorado1876, 2024-02-07:
I support this request. I'm planning a print farm and the lack of this function is holding me back from Bambu printers.

BCtopics, 2024-02-25:
This would be amazing!

DARKFiB3R, 2024-05-04:
This would be great.

Ali229, 2024-11-27:
Need this feature, this will greatly increase the production value of these machines for people that have print farms.

woodard, 2024-12-03:
I also support this feature. However another alternative workflow could be:

    When multiple plates are in the slicer, there is an option for 'Queue all plates.'
    Printer allows you to map AMS as usual for each plate.
    Printer begins printing the first plate
    Printer completes print and notifies user via handy app
    User confirms plate is cleared
    Printer asks if the print was successful, if not, asks the user if they'd like to retry print.
    < Printer prompts if filament needs to be changed, and tells the user what plate is required (textured, engineering etc.)
    > The printer continues with the next print job which can be satisfied by the loaded filament and print plate. 
    Printer starts next (or retried) print
    Repeat until all queued jobs are completed.

One way to implement this would be to have a print queue daemon bind the printer or printers. Then it exports an emulated printer and other bambu studio's bind to the print queue daemon rather than the printer itself. Most of the code to implement this probably already exists in either the printer firmware or the network protocol part of bambu studio. It would just need to be glued together differently.

ideaalab, 2025-01-10:
Not just for print farms... I usually have 5-10 Bambu Studio instances each one with a different project pending to be sent to the printer. A print queue would allow me to send the print and close the Bambu Studio.
Once a print is finished it ask you if you want to print the next one (and show you the AMS colors and materials needed on each slot).
Even when im not at the office but a college is, he could just clear the plate and continue printing without having to wait for me to arrive for the next print. I know this could be partially solved with sending plates to the SD, but its just not practical or efficient.

dazerdude, 2025-01-15:
+1, I only have the one printer, but being able to send a job to an active printer, and then "execute next" from the phone app or on the device ui would help my workflow a lot. As it stands I end up keeping open instances of the desktop app with the plates I want to print next which is awkward at best.

RobertPhotos, 2025-02-03:
I am looking for a print Queue option for bambu. 
I want to organize the order of the prints to Finnish when I am available.

RobertPhotos, 2025-11-07:
There are services that allow you have a que, for a monthly fee. I have not used those. 
It is very useful to be able to que up things. Just like a standard 2d printer. It is more critical when the print takes a long time. I am surprised that BL has not come out with one. I don't want it for a print farm. BL said that is only for print farms.

I want to be able to sort the que to adjust the print to need attention when someone is available to change the plate. Maybe do the short job next so the long print can run over night.

It would be great if it told the user what filament is expected in what slot, and what build plate is expected. But I recognize that is more complicated. Maybe just start with removed and ready to start "next".

I want to be able to tell a non-technical person to take the old plate out and put in a new, and press start next.

Would not need any cloud storage. Could be stored on the SD card.