bambulab/BambuStudio #2486: getting blank pages when i click on every setting buttons and setup wizard!

zizivivi, 2023-09-08:
### Bambu Studio Version

v01.07.04.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

CPU : i7 gen 5
RAM: 8Gb
Graphic card: Onboard

### Printer

Bambu lab X1

### How to reproduce

1. go to help > setup wizard  or  setup button in front of any options as "Flament" , " Printer" . etc...
2. you will see a blank page without any buttons and you can not able to setup anything.
[image]

### Actual results

you cant do anything!

### Expected results

you cant do anything!

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12556587/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Obermaus, 2023-09-09:
Gleicher Fehler bei mir

bambu123, 2023-09-11:
Please help check whether the Home page has landed in the dialog box is normal. Whether the problem is occasional or permanent

zizivivi, 2023-09-11:
Hi, the home page is  permanently blank too. 🙏

bambu123, 2023-09-12:
You may have encountered Chrome incompatible GPU models. You can try the following methods to solve it.

- Open environment variables settings
- Set environment variables WEBVIEW2_ADDITIONAL_BROWSER_ARGUMENTS to --disable-gpu-driver-bug-workarounds --ignore-gpu-blocklist
- Restart Bambu Studio

[image]

Tomkee, 2023-09-22:
Have the same problem. Blank page when trying to open the setup wizard. I updated my drivers for the graphic card and Set environment variables as you sed and still the same error.

bambu123, 2023-09-22:
[WebView2APISample.zip](https://github.com/bambulab/BambuStudio/files/12699287/WebView2APISample.zip)
Then this problem is troublesome. You need to use this Sample to test and open the webpage to see if it will report any errors.

bambu123, 2023-10-24:
@Tomkee Have you encountered any problems using the Sample tool? Has the tool reported any errors?