bambulab/BambuStudio #2487: FreeBSD / cmake / OpenVDB / IlmBase::Half can not be found!

cederom, 2023-09-08:
### Bambu Studio Version

git/master

### Where is the application from?

Built from source by yourself or others

### OS version

FreeBSD 13.2-RELEASE AMD64

### Additional system information

```
FreeBSD octagon 13.2-RELEASE-p2 FreeBSD 13.2-RELEASE-p2 GENERIC amd64
```

### Printer

none yet :-)

### How to reproduce

On FreeBSD:
1. `git clone https://github.com/bambulab/BambuStudio.git BambuStudio.git`.
2. `cd BambuStudio.git`.
3. `mkdir build`.
4. `cd build`
5. `cmake ..`

### Actual results

```
(...)
-- Boost::boost exists
-- TBB found in /usr/local/lib/cmake/TBB
-- Found NLopt in '/usr/local/lib'.
-- Using NLopt include directory '/usr/local/include'.
-- Using NLopt library '/usr/local/lib/libnlopt.so'.
-- OpenVDB ABI Version: 10
CMake Warning at cmake/modules/FindOpenVDB.cmake:345 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:367 (just_fail)
  CMakeLists.txt:549 (find_package)

CMake Error at CMakeLists.txt:554 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

-- Configuring incomplete, errors occurred!
```

### Expected results

CMake and build completes :-)

### Project file & Debug log uploads

```
-- OpenVDB ABI Version: 10
CMake Warning at cmake/modules/FindOpenVDB.cmake:345 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:367 (just_fail)
  CMakeLists.txt:549 (find_package)

CMake Error at CMakeLists.txt:554 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-09-12:
Hello, @cederom 

please check the wiki for the compiling guide
and make sure to follow the wiki

https://github.com/bambulab/BambuStudio/wiki

cederom, 2023-09-12:
Thanks @lanewei120 :-) That did not help but I will investigate further and will share back the results :-)

cederom, 2024-08-06:
I did some experiments but dependencies upstreams needs to be updated in the first place. Maybe one day it will work :-)

beppegeppe, 2024-10-16:
Hello @cederom.
Same update about this bug. I've gentoo system, but same issue.

cederom, 2025-08-04:
If BambuLabs could find some time to create FreeBSD Port [1] of BambuStudio that would be the best! :-) You can obtain standard installer, VM image or OCI container [2] :-)

[1] https://docs.freebsd.org/en/books/porters-handbook/book/
[2] https://www.freebsd.org/where/