bambulab/BambuStudio #2488: Linear Advance - New Tool

MartynWilliams, 2023-09-08:
I saw this linear advance tool and it may be a more elegant solution to the current tool used

https://ellis3dp.com/Pressure_Linear_Advance_Tool/

Can this be incorporated into a future version of Bambu studio

Superbobo75, 2023-09-11:
It's already there. Top Bar -> Calibration -> Pressure Advance -> PA Pattern

MartynWilliams, 2023-09-11:
I agree there's a Pressure Advance option, but I don't have those options. What version are you using, I'm on a Mac.

SaltWei, 2023-09-12:
> I agree there's a Pressure Advance option, but I don't have those particular options. What version are you using, I'm on a Mac.
> For clarity i'm referring to the current manual flow dynamics test with the calibration lines with the K factor listed to be replaced with the test described above or added to the manual test as a second option.

See https://github.com/bambulab/BambuStudio/issues/2356