bambulab/BambuStudio #2497: App becomes non responsive if left idle

matthewjpicone, 2023-09-12:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Ventura 13.5.2 (22G91)

### Additional system information

Apple M1 Max, 32gb RAM

### Printer

x1 Carbon

### How to reproduce

Open App,
Do some stuff, 
leave idle,
frozen

### Actual results

Force close app and relaunch

### Expected results

the app shouldn't freeze

### Project file & Debug log uploads

Its for all projects or even no project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

barisahmet, 2023-09-14:
Same here in Ventura 13.4 M1 Max. 

I have always left idle since 1st versions, but noticed it's freezing after some update (maybe 2 weeks ago or so).

bambu123, 2023-09-15:
可能是网络导致的临时性卡住UI线程

pedro-pedrosa, 2023-09-15:
same on ubuntu 22.04

walterwongbbl, 2023-09-25:
Hi. 
We have released a new version, please confirm this issue in the latest version.

pedro-pedrosa, 2023-09-26:
I haven't experienced this issue with the latest version yet

closed 2023-10-08 by walterwongbbl