bambulab/BambuStudio #2499: After pause, the resume does not purge filament and wipe

redsector72, 2023-09-12:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

osx 11.7.10

### Additional system information

imac 

### Printer

x1c

### How to reproduce

insert a pause in the slicer

### Actual results

After the pause, when it resume, it reheat the nozzle, purge about 1 cm of filament and does not wipe. So it takes the purge "queue" that is long about 5 cm, attached to the nozzle on the build plate.

It happens every time (I pause for inserting magnet).

### Expected results

Do a proper purge and wipe sequence.

### Project file & Debug log uploads

any project will do

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-09-12:
The pause/resume behaviour is defined by printer firmware. @sodachen22 Please have a look

angrykilobyte, 2026-02-15:
Hoping this gets fixed soon. I too am experiencing this every time after a pause.