bambulab/BambuStudio #2502: Cannot start print from SD card through Bambu Studio

molovan, 2023-09-13:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

1. Computer located NOT on the same LAN as printer
2. In Bambu Studio, go to Device - SD Card - Model
3. Hoover over model and click on print
4. click on send. 

### Actual results

Error messages says: "failed to send print job, please try again". Error code "-4030"
[image]

### Expected results

job send through to printer

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-09-25:
Hi. @molovan 
This error indicates that the current network may not be smooth. Please check the status of the network for both the studio and the printer. You can ping the printer's IP address to see if there is significant packet loss or latency. You may need to adjust the positioning of the printer.

walterwongbbl, 2023-10-23:
The issue will be closed. If you have any new issues, please create a new issue

closed 2023-10-23 by walterwongbbl