bambulab/BambuStudio #2506: Significant reduction of purged filament

GuzziRaz, 2023-09-15:
Here's an interesting tweak for reducing purge:
https://www.printables.com/model/582382-bambulab-profile-for-up-to-60-purge-reduction
https://www.reddit.com/r/BambuLab/comments/16ik8ww/new_method_to_reduce_poop_by_60_without_quality/?utm_source=share&utm_medium=web2x&context=3

Is this something you've been evaluating already? Do you see any problems with it? If not, perhaps you can improve the default profiles.

pichen2021, 2023-09-19:
Hi GuzziRaz:
    It’s a good method but it may cause nozzle clogging for some filaments(such as PLA silk), the reliability for all types and all brands of filaments in diffent printing conditions is a constraint.
Thanks.

GuzziRaz, 2023-09-19:
Sure, that is why we have filament profiles! BTW see also #2518 - this could be just another filament parameter (i.e. `cut_retract_length`) that could be set per filament (or type).

This would make Bambu filaments even more attractive as you could supply values for them (eg. 18 mm for Bambu PLA Basic, 0 for silk) while just set them to zero for all the generic ones.

barsk, 2023-11-14:
I second this feature request. Would make MMS printing so much more attainable.

Matiusauce, 2023-12-28:
Would love to see this implemented as an optional built-in feature with ability to set a retraction value. The potential waste reduction is huge.

adripo, 2025-02-20:
This feature has already been implemented for all Bambu Printers. 

For those who want to enable it:
1. Go to `Printer Settings`
2. Go to `Extruder1`
3. Enable `Long retraction when cut (experimental)`

barsk, 2025-02-20:
Could not find the setting in the latest release. Is it not released yet?

closed 2026-04-02 by GuzziRaz