bambulab/BambuStudio #2508: Bambu Studio stuck on 'print running' after print completes

wlumumba, 2023-09-15:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS bigSur 11.6.2

### Printer

Bambu Lab X1

### How to reproduce

1. Print something
2. Leave computer for 5+ minutes until computer screen goes black and print finishes
3. Try to print something else and receive 'The printer is busy with another job'
4. Check device tab in Bambu Studio and shows print in progress, even though print is finished
5. Restart bambu studio 

### Actual results

[image]
[image]

### Expected results

Printer should recognize there is no print running.

### Project file & Debug log uploads

[logs.zip](https://github.com/bambulab/BambuStudio/files/12615243/logs.zip)
[full-napkin-leg-test-2.3mf.zip](https://github.com/bambulab/BambuStudio/files/12615244/full-napkin-leg-test-2.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bobbyjball3, 2023-09-15:
Also experiencing this if my computer goes to sleep during a print. And FWIW, I'm on Windows 11

walterwongbbl, 2023-09-25:
Hi. @bobbyjball3 @wlumumba 
We have also identified this issue, as the data packets were not being pushed completely, resulting in the inability to update the print progress correctly. This issue will be addressed by the printer, and Studio has already released a new version for you to download and experience.

walterwongbbl, 2023-09-25:
@sodachen22 
打印机已知问题。

walterwongbbl, 2024-01-18:
Hi. @bobbyjball3 
Does this issue still exist after using the latest version?

bobbyjball3, 2024-01-18:
> Hi. @bobbyjball3 Does this issue still exist after using the latest version?

Seems resolved for me.