bambulab/BambuStudio #2513: Outer and Inner Wall Speeds Not Respected

jon-whit, 2023-09-15:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Big Sur 11.7.8

### Additional system information

[image]

### Printer

Bambu Lab X1C

### How to reproduce

1. Open a part
2. Change outer wall (or inner wall) speed to a low value (30 mm/s)
3. Slice the part
4. Look at the speed of the rendered slice preview

[image]

### Actual results

The sliced part preview shows a speed much greater than the outer or inner wall setting.

### Expected results

The slice part preview should not be greater than the specified outer wall or inner wall setting.

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/bambulab/BambuStudio/files/12623889/Cube.3mf.zip)

No log present after the slice.

### Checklist of files to include

- [X] Log file
- [X] Project file

jon-whit, 2023-09-15:
Relates to https://github.com/bambulab/BambuStudio/issues/1916

LiZ-Li-BBL, 2023-09-27:
Use the 3mf file you provided, speed value is normal. Could you provide the 3mf file and gcode file of which shown in your screenshot? Maybe you can check the specific value displayed at the bottom of the screen: 
[image]

LiZ-Li-BBL, 2023-10-24:
Hello,
I think the Options->Travel at the bottom of Speed Color Scheme is checked. If so, you should uncheck it and reslice, and you will get a preivew you expected. It's a bug now, we will fix it as soon as possible.

LiZ-Li-BBL, 2024-01-16:
> Hello, I think the Options->Travel at the bottom of Speed Color Scheme is checked. If so, you should uncheck it and reslice, and you will get a preivew you expected. It's a bug now, we will fix it as soon as possible.

The problem has been fixed recently. Could you please use our latest verision to check if the issue still exists? 
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.03.89

closed 2024-01-18 by SaltWei