bambulab/BambuStudio #2515: Missing wipe sequences on 0.2mm nozzle before moves during gap infill causes stringing

pedro-pedrosa, 2023-09-16:
### Bambu Studio Version

1.6.0.65

### Where is the application from?

Bambu Lab github releases

### OS version

ubuntu 22.04

### Printer

BL X1C

### How to reproduce

1. Open the attached project file
2. Slice. Enable display of travel moves so you can see the gap infill between each tooth causes a travel from one tooth to the next and there is no wipe before any of such moves

### Actual results

I get stringing which matches the travel moves of the toolhead between gap infill extrusions.
[image]

### Expected results

I injected wipe start and wipe end sequences after gap infill extrusions like the following directly in the gcode and got rid of the stringing

```
G1 X127.901 Y121.965 F30000
; FEATURE: Gap infill
; LINE_WIDTH: 0.0997943
G1 F1200
G1 X127.919 Y121.934 E.0001
G1 X128.093 Y121.871 E.0005
; LINE_WIDTH: 0.14371
G1 X128.268 Y121.808 E.00076
; LINE_WIDTH: 0.126933
G2 X128.481 Y121.722 I-1.38 J-3.726 E.00082
; WIPE_START
G1 F24000
G1 X128.268 Y121.808 E-.20998
G1 X128.093 Y121.871 E-.17002
; WIPE_END
G1 E-.02 F1800
G17
G3 Z1.06 I.929 J-.786 P1  F30000
G1 X127.971 Y120.844 Z1.06
G1 Z.66
G1 E.4 F1800
```

[image]

### Project file & Debug log uploads

[gear.zip](https://github.com/bambulab/BambuStudio/files/12640266/gear.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

pedro-pedrosa, 2023-09-16:
also in my experience with this kind of prints the exact same print presents 0 stringing with a 0.4mm and lots of stringing with a 0.2mm nozzle (not this particular model that I attached since it's so small but it's where the issue is the worst)

this happens with brand new bambu lab filaments so it's not that either

QingZhangBambu, 2023-10-10:
are you use PLA matte?

pedro-pedrosa, 2023-10-12:
these pictures were taken on a different brand of PLA. but I've had this happen on the bambulab PLA basic white color.

pedro-pedrosa, 2023-10-12:
this depends a lot on the model. this model doesn't generate any wipes but a scaled up version of it does generate wipes and does not have stringing

for some reason I cannot replicate anymore on the original file I sent (I have updated bambu studio to 1.7 in the mean time) but on this new file which I am attaching you can see it happen on the gear to the left.

[gear2.zip](https://github.com/bambulab/BambuStudio/files/12880790/gear2.zip)

for instance after slicing, on layer 10 at after the gcode line 12064 there should be a wipe sequence instead of a G1 move. the gear to the right as all the necessary wipes as far as I can tell

QingZhangBambu, 2023-10-25:
the retraction of 0.2 nozzle is not enough. we will test it.

Xiuqi-Huang-CN, 2023-10-26:
[image]
[image]
[image]
Hello. We did some tests and got some results as shown in the pictures. And my conclusion is that the filament you used were damp and not dry enough, and you can avoid the strings easily  by drying it.
As far as I am concerned, when you want to get really high qualities of the prints, you should dry the filaments before use and put them in the AMS or filament boxes that are air-tight together with some effective desiccants during the printing and storage term to protect them from the moisture in the open air. The following is the wiki about drying and protecting filaments: **_[https://wiki.bambulab.com/en/filament-acc/filament/dry-filament]_**
Besides, the filaments may get wet during the storage and transportation which could last for months between when they were produced and when they are to be printed, so, drying before use is always recommended if you want to get very high qualities.
May this can help you and wish you all the best.

Xiuqi-Huang-CN, 2023-10-26:
@pedro-pedrosa

pedro-pedrosa, 2023-10-26:
thanks for having a look and testing. I will try drying the filament thoroughly before printing and will report back here. I have noticed your tests were done with the first file I attached here, which does include wipe sequences. I will dry the filament and test again on both files that I attached here and I will report back. please keep this open until then

Xiuqi-Huang-CN, 2023-10-26:
> thanks for having a look and testing. I will try drying the filament thoroughly before printing and will report back here. I have noticed your tests were done with the first file I attached here, which does include wipe sequences. I will dry the filament and test again on both files that I attached here with 0.4mm retraction and I will report back. please keep this open until then

Ok, keep in touch these days. @pedro-pedrosa

pedro-pedrosa, 2023-10-26:
thank you so much to your attention. how do you get such sharp XY edges on this model? the gears are supposed to be around 6mm in diameter. are you scaling up the models? that will change the results. I get this issue mostly on smaller models.

anyways I have preliminary results BEFORE drying on bambulab PLA basic filament:

first model (same you printed, no scaling) - minimal microscopic stringing
[image]

second model (please see earlier post gear2.3mf) - minimal microscopic stringing on the original model and MASSIVE stringing on model to the bottom
[image]

second model with modified g-code (please download [gear-export.zip](https://github.com/bambulab/BambuStudio/files/13177058/gear-export.zip), note this is a pre-sliced file that was modified using a script, do NOT re-slice) - minimal microscopic stringing on both models
[image]

I am drying the filament right now on the printer filament dryer utility and will post back later... but as you can see there is clearly an issue with massive stringing on models that don't have wipes.

Xiuqi-Huang-CN, 2023-10-27:
@pedro-pedrosa I did scale up the gear to make the strings more visible and apparently. If you want to make the gears more detailed and delicate, you can dry the filament out before use and protect it with an air-tight box (or the AMS) and effective desiccants. Besides, set lower speeds and accelorations before you slice. The following are the recommended settings for small prints that are expected to have detailed and delicate finishes. You can refer to them and make your own adjustment according to the actual finishes. 
[image]
[image]
Wish you good luck!

pedro-pedrosa, 2023-10-27:
thank you for your attention and reply and thank you for the suggestion.

there is a clear problem in the slicer code. on most models, the slicer generates wipes (retraction + z-hop) after gap-infill extrusions before the toolhead moves through empty space. on some other models it does not, causing unusually high amounts  of stringing as you can see in the second picture I posted on my previous post. you did not replicate this in your attempts, the stringing you got was normal stringing from wet filament. I have proven the g-code can be changed to get rid of this massive stringing by creating a script that adds wipes (retraction + z-hop + circular wipe) in the aforementioned places. you can see the results in the third picture. changing speed and drying filament will get rid of the remaining stringing, but it won't solve the problem.

to print these models I am therefore forced to run scripts external to the official bambu studio or possibly I'll just switch to another slicer altogether.

thank you for the help

closed 2023-10-27 by pedro-pedrosa