bambulab/BambuStudio #2518: Ability to save flushing volumes

kr320, 2023-09-16:
As far as I know this isn't a feature and not really any way to do it besides saving a specific project which only works for that project and doesn't transfer to other projects.

Having a feature to save filament flushing volumes would save a lot of time. Rather than keeping a record, say in a spreadsheet, of all of the values from filament color X to X, a built in feature to save a profile of the volumes to transfer to other projects would be great.

One step further, saving specific color to color values. So rather than having a profile to choose from, a specific From and To color change could be saved individually.
[image]

GuzziRaz, 2023-09-18:
Perhaps also a *base* setting plus settings for *type* and *color* and add them together.

Example: We set a base flushing volume of 55, a change from black to white is +50 while the reverse is only +10, and a change from PLA to PETG is +25.
* Switching from black PLA to white PETG the used total would be 55+50+25 = 130 mm³.
* Switching from black PLA to white PLA the total would only be 55+50 = 105 mm³.
* Switching from white PLA to black PLA the total would only be 55+10 = 65 mm³.

And yes, definitively store this in some profile. Once correctly dialled in, I can't see any reason to use different values for different projects (except when we really don't care about color bleeding but that's a slightly different topic).

SaltWei, 2023-09-19:
@QingZhangBambu 记到需求池，统一开会review，不要分给我

DanTup, 2024-03-29:
This would be great - I'd love to be able to add colours to each filament profile (so I can keep "eSun Red" separate to "Bambu Red" for ex.) and then edit a big matrix that includes all of those filament/color combinations (which I can tune over time with testing). Then when selecting a filament, I can pick a profile and colour instead of just manually assigning a colour from a colour wheel.

garv3, 2024-11-09:
Still no news on this?