bambulab/BambuStudio #2521: Bambu Studio Bed incorrectly displayed after changing printer type/ctrl-z

robandwend, 2023-09-17:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 10 Enterprise

### Additional system information

 CPU: Intel(R) Core(TM) i7-10850H CPU @ 2.70GHz   
Memory 32GB
Display Nvidia

### Printer

Voron0.1->X1C

### How to reproduce

1. Open Bambu Studio
2. Select Printer with different bed size (eg Voron 0.1)
3. Add primative-cube
4. Slice the cube (as voron)
5. Switch back to prepare-tab
6. Switch to bambulabs-x1c printer
7. Clone cube - 50 copies
8. Auto-Arrange
9. CTRL-Z (twice)

### Actual results

Bed is now incorrectly displayed, with size of bambu, but buttons positions for voron0.1
[image]

Here's a recording reproducing the problem

https://github.com/bambulab/BambuStudio/assets/10753891/3505f02a-8a72-4ab1-8276-acbafedb5c41

And the log
[debug_network_Sun_Sep_17_17_08_26.log](https://github.com/bambulab/BambuStudio/files/12642493/debug_network_Sun_Sep_17_17_08_26.log)

### Expected results

Bed correctly displayed

### Project file & Debug log uploads

[beddisplayproblem.zip](https://github.com/bambulab/BambuStudio/files/12642499/beddisplayproblem.zip)
[debug_network_Sun_Sep_17_17_08_26.log](https://github.com/bambulab/BambuStudio/files/12642500/debug_network_Sun_Sep_17_17_08_26.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2023-09-18:
Thank you. We have identified and fixed this issue. The fix logic is that changed the 3D printer does not support rollback. Please look forward to version 1.8,

Haidiye00, 2024-01-16:
The issue has been fixed, please use the latest version V1.8

closed 2024-01-16 by Haidiye00