bambulab/BambuStudio #2523: Access Violation Error when generating Gcode

Minalia, 2023-09-17:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 23H2 Build 22631.2271

### Additional system information

CPU: Ryzen 7 3800x
RAM:  32GB 3600 CL14

### Printer

Bambu Labs P1P

### How to reproduce

Slice any model

### Actual results

Slicing will proceed normally and Access Violation will occur after it reaches the "Generating Gcode" step.

### Expected results

Gcode should be generated and file will be ready to transmit to printer. Application functioned properly until today.

### Project file & Debug log uploads

[Corey_Cunningham_Mask.zip](https://github.com/bambulab/BambuStudio/files/12643233/Corey_Cunningham_Mask.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12643234/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Minalia, 2023-09-17:
It's important to note that this issue only cropped up after my system lost power last night. My laptop still slices no problem.

Minalia, 2023-09-17:
Also, after doing a complete uninstall even going so far as to delete user files, the Setup Wizard pops up a black window.

bambu123, 2023-09-18:
> It's important to note that this issue only cropped up after my system lost power last night. My laptop still slices no problem.

That is to say, it cannot be reproduced?

Minalia, 2023-09-18:
It happens 100% of the time on my PC now. I'm not sure what caused it tbh.

closed 2023-09-18 by Minalia