bambulab/BambuStudio #2526: Bug on scale !

Witawat, 2023-09-18:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11

### Additional system information

cpu : i9

### Printer

X1C Combo

### How to reproduce

Because the size of the X, Y, and Z axes cannot be adjusted, the work must be scaled only. But there are problems with the workpiece.

Normal 
[image]

[image]

[image]

[image]

### Actual results

scale size 100.66%
[image]

[image]

[image]

### Expected results

Want an fw that can adjust the x y z distance or want to fix this problem?

### Project file & Debug log uploads

[New folder (2).zip](https://github.com/bambulab/BambuStudio/files/12644489/New.folder.2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-09-25:
@Witawat  Could you explain your problem in detail and the steps to reproduce it? (For example, what was your expected result, what was the actual result, and how did you encounter this problem?)

Witawat, 2023-09-25:
> @Witawat Could you explain your problem in detail and the steps to reproduce it? (For example, what was your expected result, what was the actual result, and how did you encounter this problem?)

Problems caused by I can't adjust the size of the piece to fit.
From information from various sources, bambu does not enable commands to adjust estep x y z of the machine.
and recommend methods Scale the work size up To solve this problem

[image]

But I encountered this problem with some files when the size was scaled up. Found a part that shouldn't have been there. with the workpiece as shown in the picture

I have tested with Slicer since prusa supper, no such problems, only Orca and bambu.

[image]

make me insecure Will there be excess problems with other pieces after the workpiece is scaled?

zhimin-zeng-bambulab, 2023-09-25:
Do you mean that after scaling the object, some redundant parts will appear? Could you have a try in V1.7.6 ?

Witawat, 2023-09-25:
now i record screen

https://drive.google.com/file/d/1r89ThCDQjg0o-lQ9_2UnCrJBu_O2E-ff/view?usp=sharing

zhimin-zeng-bambulab, 2023-09-25:
[image]
Do you mean that when Scale is 100.66%, a sharp corner area will appear on the top?

Witawat, 2023-09-26:
> [image] Do you mean that when Scale is 100.66%, a sharp corner area will appear on the top?

yes

zhimin-zeng-bambulab, 2023-10-08:
Since the top surface of this model has a certain inclination angle, when scaling to a specific size, this situation will occur when the cutting plane of the slice passes through this bevel.
You can cut this model at a height of 15.74, and you will find that there is this triangular area on the top of the model.

[image]

zhimin-zeng-bambulab, 2023-10-11:
This is not a BUG, ​​it is caused by the uneven top surface of the model.

closed 2023-10-11 by zhimin-zeng-bambulab