bambulab/BambuStudio #2529: X/Y Support Distance Priority

movinimage, 2023-09-18:
X/Y Support Distance Priority

When printing with soluble support is is important to use 0 as your z support distance  and x/y support distance to get good quality suppported surfaces. But, when printing using soluble support _only_ for interface layers, the x/y clearance still must be maintained so that the support structure doesn't touch the model. In this case, if an arch or filet is being supported, the x/y distance will keep the curved surface from being properly supported. Cura (and I *think* Prusa) has an option to override this special case called "Z overrides X/Y."

In Bambu Studio there is no way to override the X/Y Support Distance in cases where curved surfaces need to be supported. This makes it impossible to properly support a filet or other arch while also maintaining an X/Y support distance for the rest of the print. 

**In Cura, you can see the behavior when "Z overrides x/y" is selected. The fillet is properly supported.**
[image]

In Bambu Studio, without this option the filet is not supported. In order to support the fillet, the x/y distance must be reduced to 0, in which case the support infill will also bond to the verticals of the part, making soluble support interfaces impossible to use. Instead the entire support must be soluble which results in wasting expensive support filament and requiring an order of magnitude more filament changes.
[image]

movinimage, 2023-10-04:
Just curious if this can be added?

movinimage, 2023-10-05:
I've gotten a few private messages about this post. People are confused. I think I wasn't clear enough..so I hope this helps:

The problem is on the top steep part, not the lower part which is below 45deg. Cura has an option “Z overrides x/y.” for this exact case. it allows the support interface to actually touch the steep overhang even if it is within the horizontal separation distance. The problem is in the second example, the Bambu Studio example…where the yellow portion (the dissolvable support interface) does not actually touch the model as it must when we are getting up to 65-75-85 deg overhangs.

In Bambu studio when using a soluble support interface, with a regular support structure, you have only two options to get curved overhangs properly supported. Both are bad options, that don't solve the problem.

1- If you simply set the horizontal offset to 0 (which is the only option in Bambu Studio), then the entire support structure will touch the model, including the non-dissolvable part, and the model will be ruined.

2- Or, set a horizontal offset …like .35 which I think is default…then, the non-dissolvable part of the support will not touch the model, which is great…but the dissolvable portion of the support wont actually provide any direct support.

The only options you are left with in Bambu studio then are to use 100% dissolvable support, with a 0 horizontal offset (which results in astronomic print times and waste because of filament changes)…or model your own supports as separate parts in CAD, and then use the dissolvable filament for them.

I tried out the option in Cura…and it fixes the entire problem with 1 click.

I hope this is clearer.

movinimage, 2023-10-06:
I can see my screenshot is not from an angle that is clear. The yellow does not touch model, and the interface layers are not printed in the air. When fully zoomed in, Bambu Studio does not allow smooth small-step rotation of the view, so it is quite difficult to get a good screenshot. 

All3DP gives what might be a clearer explanation of the problem and why the setting in Cura exists. Scroll to "Distance Priority"

https://all3dp.com/2/cura-support-settings-optimize-your-supports/

The need for the override setting is a known issue on all Cura/Slicr derivatives, and is the reason the "Z ovverrides X/Y" setting exists in Cura and in Prusa I believe. I have confirmed it on screen and in printing. The only solution I've found that works is to hand-model supports in CAD and then choose dif. filaments for them in Bambu..then do a ton of customizing of settings to get a 'part' to print like a support (interface layers, sparse internals, no walls, etc...).

Here is a better screenshot so that you can see how it actually prints in Bambu. 
[image]

Without this setting available, soluble (and sometimes breakaway) support simply doesn't work in many situations unless the entire support structure is printed in the separate support material. This is not practical with a single extruder and filament changes. Best practice requires using support interfaces in the soluble/breakaway material and the rest of the support structure in the original material. This doesn't work though because the support body ends up touching the model. Only "Z overrides X/Y" fixes this issue by allowing the support body to keep a safe distance while at the same time allowing the interface layers to make full z contact with the model to be supported.

Please consider bringing this setting to Bambu Studio from Cura.

movinimage, 2023-10-06:
Another...maybe clearer explanation than mine:

See here and scroll to "change support distance priority"

https://3dprinterly.com/how-to-fix-cura-supports-not-touching-model-or-print/#:~:text=Another%20strong%20recommendation%20from%20Cura%20users%20is%20to,affecting%20the%20set%20Z%20distance%20to%20the%20overhang.

tangketan, 2024-01-03:
Sounds like a reasonable feature request. We'll look into it.

kvnper, 2024-02-29:
This setting is sorely needed. I've played with many settings to try and combat this but they're always at the cost of something else

movinimage, 2024-02-29:
That would be awesome. It would go a long way towards making soluble/breakaway support workable for overhangs.

radekkpl, 2024-03-04:
What about to only add x/y distance settings for interface layer? So you can keep normal support distant from part and only interface layer set x/y and Z distance to 0 when printing with solutable support? But in this case i discovered another problem and that is, that support structure shape is updated on every other layer, not each layer. (I will use later terms support layer generate mode, support layer copyied mode) So Z distance is culminating between 0 and one layer height. And most important part is, sometimes the top layer of support is not even generated and you have one lyaer gap. I checked this case and it happens when new normal layer is in place where support will be in layer copied mode, so previous layer is not generated.

movinimage, 2024-03-04:
The 'Z overrides x/y' only does something in situations when the x/y distance setting is preventing Z the required z contact.  I *think* that what you are asking for is exactly what this setting does...It evaluates on a per-layer basis I believe, so that they x/y is only modified when it keeps Z from touching as it should, and since interface layers are the only layers that are (should) configured to touch, they are already the only ones likely to be affected.

radekkpl, 2024-03-07:
Yes. But Cura use another backend slicer, than Prusa/Bambu/Orca so this Z overide can be missed or hard to implement. But X/Y distance for Interface layer is most easy way to implement i think, because during generation supports, that settings is already used. But XY distance use same value for normal support and interface support as well. So maybe quick hack would be go this way until proper solution is used. But i do not know the code, so this may be also hard to implement if interface layer is generated directly from normal support shape. But support generation is weak spot in every slicer.

movinimage, 2024-03-07:
Cura & Prusa/Bambu/Orca are have slicing engines originally based on Slic3r/Superslicer/Prusaslicer. That is part of the reason that they remain (mostly) open source. There are certainly modifications. AS the CEO of Bambu Lab said it:

https://www.reddit.com/r/BambuLab/comments/v07e6v/to_open_or_not_to_open_that_is_the_question/

1. A slicer that is based on Slic3[r/Superslicer](https://www.reddit.com/r/Superslicer/)/Prusaslicer; we heavily modified it to include some new algorithms and tricks.

My GUESS is that it would be easiest to implement the original override code, but I am not familiar enough with the codebase to know :)

kvnper, 2024-03-11:
AFAIK Cura is not based on slic3r or its variants.