bambulab/BambuStudio #2532: Split parts with different colors of the same filament fail to carry over to the slicer - ie. the slicer just thinks they are all the same filament/color.

ianhill60, 2023-09-19:
### Bambu Studio Version

all versions since around 1.5.0.60

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Mac Ventura 13.5.1 and Windows 11 Pro 22H2

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Export any .stl with multiple bodies, split the object to parts in Bambu Studio, change the filament type of some of the parts (I used two different colors of polycarbonate), they will appear fine in the prepare tab, press slice, navigate to the preview tab. 

### Actual results

The slicer preview fails to differentiate between unique filaments of different parts in the same object. Slices it as one continuous filament. 

### Expected results

The object should be split into regions corresponding to the different filament colors. 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-10-07:
@ianhill60  Could you share your 3mf file?

ianhill60, 2023-10-16:
For sure. Please find attached. 
[InsertPrintSolid.3mf.zip](https://github.com/bambulab/BambuStudio/files/12920725/InsertPrintSolid.3mf.zip)

zhimin-zeng-bambulab, 2023-10-17:
@ianhill60 
When different parts under the same object overlap each other, the overlapping part will take priority in using the lower part in the objectlist.

So, if you want to display other colors, you can drag the "3DPrintedAntEnclosure.STL_28" part to the front of the objet list. As shown below
[image]

ianhill60, 2023-10-17:
Incredibly helpful. Thanks a ton.

closed 2023-10-17 by ianhill60