bambulab/BambuStudio #2533: Can't print in PLA Aero using AMS

tatertacoma, 2023-09-19:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C with AMS

### How to reproduce

Load PLA Aero into the AMS (any slot)
Wait for AMS to do its thing
Make sure AMS is synced with Bambu Studio
Set object in a project to use PLA Aero (selected from synced filament by right-clicking on object and selecting the filament under "Change Filament"
Slice object and Select the Print button

It appears in any/all project files I open/create and it doesn't matter if I restart the printer or reload the filament with the AMS.
I have only noticed this happening with PLA Aero with everything else I have being PLA.

### Actual results

In the final menu before sending it off to the printer it does not allow me to continue because it says "Please click each filament about to specify its mapping AMS slot before sending the print job" and if I click on the PLA-AERO button to select the AMS slot (which it doesn't show a slot selected just "PLA-AERO" with a "-" under it) it doesn't allow me to select a slot and says "Note: Only the AMS slots loaded with the same material type can be selected" and it does not allow me to send the print.

### Expected results

I would think that I would continue normally like every other filament that I have used by automatically selecting the AMS slot and just allowing me to set the settings I want (Timelapse/Bed leveling/Flow Dynamics Cal.) and being able to just send the print.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12655662/log.zip)
[Simple Indoor Glider V2.zip](https://github.com/bambulab/BambuStudio/files/12655669/Simple.Indoor.Glider.V2.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-09-20:
Hi. @tatertacoma 
We will confirm the issue

tatertacoma, 2023-09-22:
Still happens after updating firmware...
[image]

walterwongbbl, 2023-09-25:
Hi. @tatertacoma 
We believe that PLA-AERO is not consistent with other PLA materials.
If you want to continue printing, please modify the filament preset to PLA Basic

[image]

JohnMarzulli, 2023-10-06:
Found this bug due to experiencing the same issue.

The expectation as a user is that the system would warn me if a plastic was not compatible with the AMS.

The error about being unable to contact the printer is erroneous and incongruent with the Bambu Studio experience.

Having to disconnect the AMS and then feed the PLA Aero through the bowden tube is another horrible experience. Why would you include the NFC chip if this is what you expect users to do?

JohnMarzulli, 2023-10-06:
The product page states that PLA Aero is AMS compatible.

The "Bug" tag needs to be returned.

[image]

walterwongbbl, 2023-10-09:
Hi. @tamasmeszaros @JohnMarzulli 
We are very sorry to inform you that we have identified this issue due to a configuration error that resulted in a material type mismatch. We have fixed this issue and will include this change in future versions.

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl