bambulab/BambuStudio #2534: All log files contain secrets and are unsuitable for public submission to github

FyrbyAdditive, 2023-09-19:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Ventura 13.5.2 macOS Sonoma RC

### Additional system information

Macbook Pro 16" 2021 M1 Pro, 32GB RAM

### Printer

X1C

### How to reproduce

Upload a log file to github and secrets to access your models (network debug log) and camera (standard debug log) are leaked. I have checked the AWS keys in the log files from august to september and they are not rotated and nor do they have a short lifespan.

There is no warning of this in this bug tracking system where the logs are requested.

This is a massive security issue for anyone that has already submitted any tickets containing any such logs.

### Actual results

Logging excessive secrets and requiring these logs as part of bug reports, which are public, pushes users that are trying to be helpful and improve your products by logging tickets into a serious security problem if they are printing anything private or commercially sensitive.

There is also the problem that this also may or may not give public access to cameras in devices in peoples homes.

### Expected results

All secrets for all users that have submitted tickets to Github need to be rotated for everyones security.

An investigation into potential abuse of these secrets needs to be conducted.

Probably consider redacting the excessive logging of secrets from log files so that it is possible for users to submit bugs securely and without having to worry.

This will likely spark a flood of GDPR requests if it is picked up by the media or other users.

### Project file & Debug log uploads

Not uploading log files for obvious reasons.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FyrbyAdditive, 2023-09-19:
It looks like many and possibly all users share the same AWS S3 bucket, so one also has to wonder how the permissions are organised in this bucket regarding AWS keys - can the keys from one network debug log file be used to access all users models sharing the same bucket?

FyrbyAdditive, 2023-09-19:
It looks like the keys are valid for 24 hours, hence any tickets submitted in this timeframe after printing a model could be assumed to be potentially abused.

lanewei120, 2023-09-19:
@FyrbyAdditive 

hello, we have checked this logic in the log,  and confirmed the camera url is just a temporary url, 
it will be invalidated in serveral minutes after your camera disconnected from Bambu Studio

we are sorry bring you into this worries, 
and we will remove this information (mainly for debug) in next coming version

also we will add the encryption of our log in the future

FyrbyAdditive, 2023-09-19:
Thank you for the confirmation on the camera. It sounds like the AWS keys in the other log file are problematic however, so it may be nice to put something (wait 24h) in the bug submission form as a preventative measure.

lanewei120, 2023-09-19:
> Thank you for the confirmation on the camera. It sounds like the AWS keys in the other log file are problematic however, so it may be nice to put something (wait 24h) in the bug submission form as a preventative measure until the other controls you mention are added.

I am sorry currently I can not understand the AWS key you mean
do you mean the aws url in network log?
such as 
`[2023-09-19 22:01:51.620993 +08:00] [I] [T 32472]: Tracking get_url = {"message":"success","code":null,"error":null,"host":"user","urls":[{"type":"event_track/slicer","file":"2444659047_track_01.07.06.03_20230919211745_679f335bb024b918c9e032f30e1a3bda1695128877.zip.enc","url":"https://s3.us-west-1.amazonaws.com/bbl-prod-us-upload/users/1959608957/event_track/slicer/20230919140150.81/2444659047_track_01.07.06.03_20230919211745_679f335bb024b918c9e032f30e1a3bda1695128877.zip.enc?AWSAccessKeyId=AKIAXY6FH2ERIPQ7LRNA&Signature=gBxcsMvksqkMQno%2Bslvw4me75LU%3D&Expires=1695218510"}]}`

if yes, it is just a upload-only url, which can not be used for downloading
no other sensitive information

anyway, we will also add the encryption of such log in the future

FyrbyAdditive, 2023-09-19:
Thank you again for this further clarification, will now close this ticket.

closed 2023-09-19 by FyrbyAdditive