bambulab/BambuStudio #2535: Ubuntu Linux v1.7.4 keep crashing on webcam streaming

Dario-Ciceri, 2023-09-19:
### Bambu Studio Version

1.7.4

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04.3 LTS

### Printer

Bambu Lab X1C

### How to reproduce

Download [gstreamer](https://gstreamer.freedesktop.org/documentation/installing/on-linux.html?gi-language=c), open BambuStudio, go to device and click on streaming button (play button) and it crashes....

### Actual results

crashes

### Expected results

should just works fine...

### Project file & Debug log uploads

no project file, just log
[log.zip](https://github.com/bambulab/BambuStudio/files/12659112/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-10-16:
@Dario-Ciceri 
Hello, sorry we have missed this issue before
do you still have this issue, if yes,

could you have a try with the newest [ubuntu image ](https://github.com/bambulab/BambuStudio/releases/download/v01.07.07.89/Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage)

and I just built a[ docker image](https://github.com/bambulab/BambuStudio/suites/17263526733/artifacts/986370117), could you also try this image?

Dario-Ciceri, 2024-03-04:
> @Dario-Ciceri Hello, sorry we have missed this issue before do you still have this issue, if yes,
>
> could you have a try with the newest [ubuntu image [...]

Hi, sorry for the delay in answering only now, unfortunately I have no way to test anything else on linux due to some hardware problems with the pc and I had to format everything but at the moment I have no way to get back into it, sorry.

closed 2024-03-04 by Dario-Ciceri