bambulab/BambuStudio #2538: Filament profiles disappear in BambuStudio

chippymiller, 2023-09-19:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

13.4 (22F66)

### Additional system information

When switching to a 0.2mm nozzle in BambuStudio, all of the preset filament profiles disappear. All the profiles stay when switching to any other size nozzle.

### Printer

X1C

### How to reproduce

Go to BambuStudio
Click on printer preset
select .2mm nozzle preset
go to filament presets
only stock presets show up

### Actual results

Saved presets don't populate.
Tried to log out then log back in.
Tried shutting printer down, then back on. 

### Expected results

It should show the saved filament profiles like it does for all the other nozzle sizes/printer profiles.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12666426/log.zip)
[bambustidio zip file.3mf.zip](https://github.com/bambulab/BambuStudio/files/12666432/bambustidio.zip.file.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-09-25:
Are you referring to the disappearance of user-defined presets?

suit, 2024-01-14:
@bambu123 i ran into the same issue, some profiles disappear when selecting the 0,2 mm nozzle

all user profiles disappear and also the Generic TPU profile is not available anymore

i'm aware that TPU lists as "not recommended" in the filament profile list, but yet the default profile just vanishes
https://wiki.bambulab.com/en/x1/manual/compatibility-and-parameter-settings-of-filaments