bambulab/BambuStudio #2539: Filament backup from external spool holder to AMS

GuzziRaz, 2023-09-20:
It should be possible to start a print from the external spool, then once it runs out continue from an AMS spool - just like AMS-only filament backup.

There is no technical problem hiding here, we merely need the software to allow it: Once you detect the external spool's filament ran out, the PTFE tubes are already empty so you can load next spool from AMS and continue with no user interaction.

motley74, 2023-12-16:
I would like to vote for this feature as well. I have a splitter on my input PTFE tube that allows me to load external spool without disconnecting AMS. With this in place, it should be possible for the AMS to load filament from AMS when the external spool runs out. May require a warning to the user to ensure that both AMS and external spool are connected via a splitter but other than that this should be possible.

vivalton, 2025-02-17:
Feels like many users would like this kind of feature to be added. Any comment from Bambu Lab team?