bambulab/BambuStudio #2541: broken project file, can not loading model file

PandaBalls, 2023-09-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

win10 pro

### Printer

X1C

### How to reproduce

1. save project file with several plates
2. close studio 
3. reopen studio and this project file
4. somethings it said "loading of model file failed"
5. then I lost my project file...

### Actual results

[image]

### Expected results

open my project file

### Project file & Debug log uploads

[Dyna.zip](https://github.com/bambulab/BambuStudio/files/12668693/Dyna.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-09-23:
@PandaBalls 

hello, it seems some of the model object's name is wrong after saving

what is the original name of the step file(such as "Down_73mm_Adj-磁1...")
and did you modify this name by yourself to anyother names?

meanwhile, I just repaired this file for you
[Dyna_new.zip](https://github.com/bambulab/BambuStudio/files/12705802/Dyna_new.zip)

PandaBalls, 2023-09-27:
> @PandaBalls
>
> hello, it seems some of the model object's name is wrong after saving
>
> what is the original name of the step file(such as "Down_73mm_Adj-磁1...") and did you modify this name by yourself [...]

I didn't modify the step name. and I downloaded the file you send me, I find that one step name have been changed: Down_73mm_Adj-磁1-2. the original name is "Down_73mm_Adj-磁12". 
Maybe I should use another name for this object? Do you think it was broken cause of step name?

lanewei120, 2023-09-27:
> I didn't modify the step name. and I downloaded the file you send me, I find that one step name have been changed: Down_73mm_Adj-磁1-2. the original name is "Down_73mm_Adj-磁12". Maybe I should use [...]

yes, it is related with the object name

however, we don't know how to reproduce it,
could you send this step file with the original name to us for a check?
if we can reproduce here ,we can have more dig into the root causes

thanks

PandaBalls, 2023-09-29:
I have changed the original stp file's name, I will try new name for a while, find if it works well. I'll let you know if prj file broke again, thank you for your support.

lanewei120, 2023-11-24:
close this issue as not reproduced, and solved

closed 2023-11-24 by lanewei120

PandaBalls, 2023-12-01:
another Prj file broke again..

PandaBalls, 2023-12-01:
Could you let me know how to fix the broken project file?

PandaBalls, 2023-12-01:
> @PandaBalls
>
> hello, it seems some of the model object's name is wrong after saving
>
> what is the original name of the step file(such as "Down_73mm_Adj-磁1...") and did you modify this name by yourself [...]

Could you let me know how to fix the broken project file?

lanewei120, 2023-12-01:
I can not remember the extract steps,
just like thiis:
the 3mf is a zip indeed, extract it, and find the invalid chars in 3D/3dmodel.model
and correct them
then re-zip it

PandaBalls, 2023-12-01:
> I can not remember the extract steps, just like thiis: the 3mf is a zip indeed, extract it, and find the invalid chars in 3D/3dmodel.model and correct them then re-zip it

So you didn't know which model's name cause the problem, try to guess?

PandaBalls, 2023-12-01:
> @PandaBalls
>
> hello, it seems some of the model object's name is wrong after saving
>
> what is the original name of the step file(such as "Down_73mm_Adj-磁1...") and did you modify this name by yourself [...]

Which is the invalid char in this model's name?

PandaBalls, 2023-12-01:
> I can not remember the extract steps, just like thiis: the 3mf is a zip indeed, extract it, and find the invalid chars in 3D/3dmodel.model and correct them then re-zip it

Could you help to repair this project file? I tried the method you told me, But can not find the invalid chars to fix it.

lanewei120, 2023-12-01:
Sure, Where is the file?
please upload here, I can repair it for you

PandaBalls, 2023-12-01:
> Sure, Where is the file? please upload here, I can repair it for you

Here is the broken project file:
https://we.tl/t-bVhBZxx5Hi

lanewei120, 2023-12-01:
please find the file here
https://www.dropbox.com/scl/fi/jpyskr3120usz202rvzti/53mm.3mf?rlkey=q154x7hpx73p07ky1xzw6riro&dl=0

lanewei120, 2023-12-01:
there is a line end here, which is most likely caused by invalid char
[image]

I just modified following files:
3dmodel.model
3dmodel.model.rels
53-磁环-_无胶_470.model

lanewei120, 2023-12-05:
> > there is a line end here, which is most likely caused by invalid char ![image](https://private-user-images.githubusercontent.com/106916061/287255764-5b2494b7-1a96-40aa-aca4-a466095442c7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3MDIwNzQsIm5iZiI6MTcwMTcwMTc3NCwicGF0aCI6Ii8xMDY5MTYwNjEvMjg3MjU1NzY0LTViMjQ5NGI3LTFhOTYtNDBhYS1hY2E0LWE0NjYwOTU0NDJjNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIwNFQxNDU2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YjMxYTQwNmM5YTg0YjY1YThkM2IxYWE2MTc5ODhmYzUxOTY3MTNlODc5NTBmZWUwNTNmMzE4Njc1YzVhZjgxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TsV3r7-Nw0UY_3aT6MtGo7Hcjfxxn8TInKhrha_XR14)
> > I just modified following files: 3dmodel.model 3dmodel.model.rels 53-磁环-_无胶_470.model
> 
> Thank you so much, I just successfully repaired it with your tips.

could you show us how to reproduce it?

we need to avoid this error in your case.
and could you also share the file which causes this error?

thanks