bambulab/BambuStudio #2545: Cannot unset default color in filament

fangly, 2023-09-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

P1P

### How to reproduce

1. In the Prepare tab, select a filament.
2. Click on the edit button
3. Observe that the default color is undefined.
4. Set a default color.
5. There is no way to reset the default color, i.e. to set it to undefined again.
[image]

### Actual results

There is no way to reset the default color, i.e. to set it to undefined.

### Expected results

There should be a way to set the default color to its initial value, i.e. undefined.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-09-25:
Maybe relate to Linux version.

bambu123, 2023-10-07:
I know the problem you're talking about. The color dialog box does not support selecting invalid colors. We may need to add a button to reset to the unavailable value.

Chewieez, 2024-09-01:
This is still an issue in the latest Bambu Studio on MacOS vr 1.9.5.51

I had to manually locate the .json file for the filament profile and erase the lines for setting the default color.