bambulab/BambuStudio #2546: Can't change the K Value in Bambustudio (German language)

Mastaila, 2023-09-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS Ventura 13.5.2

### Additional system information

M1 Mac mini

### Printer

Bambulab P1P

### How to reproduce

While the Bambulab Studio is in the German language I can't change the k value of the ext spool. I tried 0,025 and 0.025 but it won't change. As soon as I change the language to English, I can change the value without any issues. 

### Actual results

K value won't change in German language (Bambustudio)

### Expected results

It should change ? :)

### Project file & Debug log uploads

No need

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-09-25:
[image]
@Mastaila  Do you mean that the k value cannot be changed?  Could you try again with the latest version(V1.7.6)?

zhimin-zeng-bambulab, 2023-10-07:
@Mastaila  Could you have a try with the latest version(V1.7.6)?

zhimin-zeng-bambulab, 2024-01-16:
This is a BUG related to minority languages, but we have not yet found a good way to solve this problem. We will continue to try to solve it in the future.

closed 2024-08-20 by zhimin-zeng-bambulab