bambulab/BambuStudio #2547: Manual Flow Dynamics Calibration NOT working

APLOKOR, 2023-09-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 64-bit (10.0, Build 22621) 

### Additional system information

 Processor: 12th Gen Intel(R) Core(TM) i5-12500H (16 CPUs), ~2.5GHz
Memory: 32768MB RAM
Intel(R) Iris(R) Xe Graphics
NVIDIA GeForce RTX 3060 Laptop GPU

### Printer

P1S

### How to reproduce

Calibration>Flow Dynamics OR Flow rate

### Actual results

[image]

when finish Calibration printing, cant click next button
and I cant fix value on this page

### Expected results

.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12673315/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-09-25:
@APLOKOR  There are too many files in this log folder. Could you tell me the approximate time period when this problem occurred?

zhimin-zeng-bambulab, 2023-09-26:
@APLOKOR  You can have a try in V1.7.6, Manual Flow Dynamics Calibration has improved.

APLOKOR, 2023-10-02:
@zhimin-zeng-bambulab 
[image]

My reply is late I'm sorry.
This bug is still in the same state as we did the latest update.
Both calibrations are have same problem.
I owe you for your services.

zhimin-zeng-bambulab, 2023-10-07:
@APLOKOR  Could you share your new log file, when you send Manual Flow Dynamics Calibration?

APLOKOR, 2023-10-15:
@zhimin-zeng-bambulab 
It may be my personal mistake, but I experienced an additional communication anomaly today.
During the filament replacement, the printer is out of control, and when I reconnect from the device list, the backlog command is executed at once.

G-code is also abnormal, it Work in the air and Occurs excessive motor movement to points outside the available range

https://youtu.be/8GuqxyvsS0g
https://youtu.be/PyPCJKJ1YmA
https://youtu.be/lfx4ICPWlHg

I hope it helps to fix the bug

[log (2).zip](https://github.com/bambulab/BambuStudio/files/12909746/log.2.zip)

zhimin-zeng-bambulab, 2023-10-19:
@APLOKOR  Could you reproduce it again and send us the Studio log and printer log together? And record the approximate time when you reproduce the problem. Thanks！

zhimin-zeng-bambulab, 2024-01-18:
Could you have a try with firmware 01.05.01, it has improved this kind of problem.

zhimin-zeng-bambulab, 2024-02-01:
@APLOKOR  Does this problem still exist with firmware 01.05.01?

closed 2024-06-26 by zhimin-zeng-bambulab