bambulab/BambuStudio #2550: Camera freezes regulary since update

AroundThisWorldAgain, 2023-09-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

OSX 13.5.2

### Additional system information

M2 Macbook Air

### Printer

X1C

### How to reproduce

Lan mode, whenever I click the "play" button to start the camera it freezes after almost exactly 10 seconds.  Every single time.  Worked fine prior to recent software & firmware updates.

This is basically a clone of issue #2150, which was "fixed" in Aug.

### Actual results

Camera freezes

### Expected results

Camera should not freeze

### Project file & Debug log uploads

[Screw Hole Finders.3mf.zip](https://github.com/bambulab/BambuStudio/files/12676459/Screw.Hole.Finders.3mf.zip)
[debug logs.zip](https://github.com/bambulab/BambuStudio/files/12676460/debug.logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Nidget67, 2023-09-23:
My Win 11 PC does the same. I print from Fusion 360 and it opens Bambu studio with a blank model and then opens up the F360 export in another instance. My model crashes when viewing the device print, but in the background is the empty Bambu studio playing the video without any problem.

bambu123, 2023-09-25:
From the logs, it can be seen that you are using the LAN Mode Liveview of the printer, and the LAN network may be unstable. You can try restarting the printer first to troubleshoot the stability issue of the printer LAN Mode Liveview.

AroundThisWorldAgain, 2023-09-25:
I also said I was using lan mode in the issue 😉

The network is fine, the printer has been rebooted, it's almost right next to the router, and as mentioned, this worked perfectly right before I updated the software & firmware.

bambu123, 2023-10-24:
> I also said I was using lan mode in the issue 😉
> 
> The network is fine, the printer has been rebooted, it's almost right next to the router, and as mentioned, this worked perfectly right before I updated the software & firmware. A network issue also doesn't explain it consistently failing after exactly 10 seconds over and over and over.

If the issue persisted, please create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).

erpalma, 2024-01-11:
Same here, video locks up after a few seconds. Version 1.8.3 , Ubuntu 22.04. The feed works fine using `ffmpeg.sdp` on ffplay, so I guess this is a BambuStudio issue.

SaltWei, 2024-01-18:
If the issue persisted, could someone help to upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) so can be further analysed?

AroundThisWorldAgain, 2024-01-18:
Um, no?  I uploaded logs when I created this issue 4 months ago and you ignored them.

bambu123, 2024-01-18:
> Um, no? I uploaded logs when I created this issue 4 months ago, and again and month later when you asked, and you ignored them both.

Did you create a ticket to the X1 printer team?
If yes, please shared the ticket link here.

bambu123, 2024-01-18:
> I also said I was using lan mode in the issue 😉
>
> The network is fine, the printer has been rebooted, it's almost right next to the router, and as mentioned, this worked perfectly right before I [...]

Please use other players that support RTSP (such as ffplay) to play the following URL directly. See if it can play normally for a long time.
rtsps://bblp:<access-code>@<printer-ip>/streaming/live/1

AroundThisWorldAgain, 2024-01-18:
No.  I'm not going back to repro an issue you should have addressed when I opened the issue in September.  And October.

If someone else wants to deal with your miserably pathetic support they can.  I'm out.

closed 2024-01-18 by AroundThisWorldAgain

bambu123, 2024-01-19:
I think this is a printer-related issue, so I need a printer log.
could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).