bambulab/BambuStudio #2554: Triggering manual flow dynamics calibration causes p1s to Lock up and reboot

areoseek, 2023-09-20:
### Bambu Studio Version

1.74.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

CPU: AMD ryzen 5800x 
Ram: 32gb DDR4 3666mhz
printer version: 1.04
(p1s and computer running bambu studio are on same wifi network)

### Printer

bambu p1s

### How to reproduce

open bambu studio
click the calibration tab
click flow dynamics
select 0.4 nozzle
select bambu asa profile
select textured pei plate
click calibrate 

### Actual results

printer locks up and crashes almost immediately, bambu studio looses communication with printer and disiplays "N/A" 
[image]
[image]

### Expected results

printer should begin calibration print

### Project file & Debug log uploads

[debug_Wed_Sep_20_16_51_57_7684.log.zip](https://github.com/bambulab/BambuStudio/files/12677498/debug_Wed_Sep_20_16_51_57_7684.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

areoseek, 2023-09-20:
[image]
(printer does this as soon as I click calibrate)

areoseek, 2023-09-20:
This issue DOES NOT happen in "Lan only mode"

Arty40912, 2023-09-22:
i have this problem to!!

sodachen22, 2023-09-23:
Could someone create a ticket and upload the log? https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log