bambulab/BambuStudio #2555: "Send print job to" Dialogue Defaults To Incorrect AMS Filament Slots vs Project Settings For Support Interface

KHeidolph, 2023-09-20:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

MS Surface Book 3

### Printer

Bambu Lab X1 Carbon w/ AMS

### How to reproduce

1. Load White PETG into AMS 1 and 2.  Load Black PLA into AMS 3 and 4.
2. Create a project requiring supports.
3. Set project and supports to print from AMS 2 (PETG)
4. Set support interface to print from AMS 4 (PLA)
5. Click on print plate.

### Actual results

"Send print job to" dialogue box indicates AMS 2 (PETG) and AMS 3 (PLA) will be used. 

[image]

### Expected results

"Send print job to" dialogue box indicates AMS 2 (PETG) and AMS 4 (PLA) will be used. 

***Note that even though AMS 4 (PLA) was selected as the filament to use for support interface in the project configuration, the "Send print job to" dialogue box set it to AMS 3 (PLA).  The "Send print job to" dialogue should use the same filaments indicated in the project settings by default. 

### Project file & Debug log uploads

[Bambu Lab Support Ticket Submission_20230920_1.zip](https://github.com/bambulab/BambuStudio/files/12677549/Bambu.Lab.Support.Ticket.Submission_20230920_1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bam-winterfell, 2023-09-25:
AMS matching will prioritize matching slots with consistent colors and materials. When the materials of the two slots in AMS are the same, the slot with the smaller serial number will be matched. If you want to use materials from another slot, please select it manually.
Wishing you a happy life

bam-winterfell, 2023-09-25:
Hi. @KHeidolph 
AMS matching will prioritize matching slots with consistent colors and materials. When the materials of the two slots in AMS are the same, the slot with the smaller serial number will be matched. If you want to use materials from another slot, please select it manually.
Wishing you a happy life

closed 2023-11-14 by bam-winterfell