bambulab/BambuStudio #2562: Control Chamber fan in studio.

ebourlet, 2023-09-21:
**Is your feature request related to a problem? Please describe.**
Why can't I control the chamber fan via the filament profiles?
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to be able to control the chamber fan via Bambu Studio. I like to run it on high when printing PLA and run it at 60% when running ASA

SaltWei, 2023-09-23:
This is under developing

gustafssone, 2023-10-10:
I can confirm I have the same issue. It works with Machine G-code but not with filament g-code.
; Filament gcode
M106 P3 S50 ; Run chamber fan at 20%

Machine start G-Code:
;==CUSTOM for ABS/ASA
{if (filament_type[initial_no_support_extruder]=="ABS")||(filament_type[initial_no_support_extruder]=="ASA")}
         M106 P3 S25 ; Run chamber fan at 10%
{else}
         M106 P3 S0 ; turn off chamber fan
{endif}  ; exhaust VOCs from chamber for ABS/ASA filament

This will set the chamber fan to only 10%, meaning filament g-code doesnt work (I checked the g-code, and it is there, just not working).

I have a P1S.