bambulab/BambuStudio #2565: A floating support

Arty40912, 2023-09-22:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU : 11th Gen Core i5-11400H @2.70GHz
GPU : RTX 3060 (laptop)
Ram : 32 GB
Storage : 1 TB

### Printer

Bambu lab p1p

### How to reproduce

upload the model stl file that i send and using the 0.28 layer high system preset setting and click to gen the support and slice it

### Actual results

I upload the file in that picture and its have the bug that its have a floating support on the model when i slice it on the program.

[image]

### Expected results

its should not generate the floating support on that model

### Project file & Debug log uploads

[Lead lock.zip](https://github.com/bambulab/BambuStudio/files/12697037/Lead.lock.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12697048/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2023-10-24:
Thanks for reporting the issue. We'll fix it in the next version.

tangketan, 2024-03-08:
Sorry, it turns out fixing this issue is quite complicated and requires a big modification of normal support, which is not acceptable since tree support is the future. Therefore, we've decided not to do it. Please try tree support.