bambulab/BambuStudio #2566: Bambu Studio software is crashing when pressing calibrate

Kroella, 2023-09-22:
### Bambu Studio Version

V1.7.6

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel 13900K
Ram: 64GB
GPU: Gigabyte RTX 4090

### Printer

Bambu Lab P1P

### How to reproduce

Clicking calibration in the main window

### Actual results

Crashes  the software to desktop

### Expected results

You should enter the calibration page

### Project file & Debug log uploads

[crash_Fri_Sep_22_12_34_41_0_done.log](https://github.com/bambulab/BambuStudio/files/12699854/crash_Fri_Sep_22_12_34_41_0_done.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-09-23:
@Kroella 

thanks for reporting this issue

could you also help to upload the file *.dmp and *.log in the same log directory?

zhimin-zeng-bambulab, 2023-10-11:
@Kroella  Does this problem still exist with the latest version? If it still exists, could you provide the log file and dump file?

zhimin-zeng-bambulab, 2024-01-18:
This problem cannot be reproduced and the user has not responded for a long time. Close it.

closed 2024-01-18 by zhimin-zeng-bambulab

LukasSlouka, 2024-02-03:
I had this happen to me as well. It happened after I changed nozzle from 0.4 to 0.2 on my P1S. I immediatelly wanted to calibrate. I have switched the preset to the 0.2 nozzle and pressed calibrate. Then chose the correct settings for calibration and when I tried to send it, it either crashed the application or it provided error message, that the memorized nozzle is still 0.4. I spent some time on this but got nowhere, so decided to print something small to see how the printer does. At that point, when I clicked send to printer button, it showed another window saying that the memorized nozzle is different but this time it gave me option along the lines of "configure and send to printer". I did that, the print started and finished without issues. Afterwards I was able to run calibration without any errors or crashes, so the issue was definitely with the memorized nozzle settings, whatever that means.