bambulab/BambuStudio #2567: Setup wizard show blank page

Tomkee, 2023-09-22:
### Bambu Studio Version

1.7.6.91

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

GPU 1070 (latest drivers)

### Printer

P1P

### How to reproduce

Help
Setup Wizard
Blank Page

### Actual results

Blank Page

### Expected results

Settings
[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-09-23:
@Tomkee 

is it a 100% produciable issue?
and did it work before?

could you help to upload the log for us to have a check?
just follow this guide https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

thanks

SnickCH, 2023-10-14:
I have the same issue on Linux

System: 64bit Garuda Linux (KDE Version)
Bambustudio:
- Fedora AppImage (1.7.7.8.9)
- chaotic-AUR (1.7.7.88-1)

Additional information I found on the net (I can't verify it)

https://aur.archlinux.org/packages/bambustudio
- https://aur.archlinux.org/packages/bambustudio-bin
- ->georgewoodall82 commented on [2023-08-19 14:03 (UTC)](https://aur.archlinux.org/packages/bambustudio-bin#comment-930010): " I recommend a light GTK theme for BambuStudio, as I had problems with text being invisible with a dark GTK theme. You can do this with for example GTK_THEME=Adwaita bambu-studio. Also, if you have an nvidia card and the setup wizard is blank, what worked for me is using wayland."

SnickCH, 2023-10-14:
I just tested it with differente themes (light) for KDE. It looks the same: The wizard is still "blank" (and grey). I don't know if I have to change other things too to verify it. In other applications the light theme is used as configured.

SnickCH, 2023-10-14:
@lanewei120 I managed to use the wizard (blank) pressing more or less where the buttons should be
https://wiki.bambulab.com/en/software/bambu-studio/studio-quick-start

(I don't know exactly what I choose but the nework plugin was installed). This was just as a test. I will remove it again (because I don't now what regions etc. I set)

Therefore it is just a problem of how it is presented. I hope that helps the investigation.

lanewei120, 2023-10-15:
@SnickCH 

thanks for doing this

the homepage and the setup wizard page are web pages
so they are related with web engine used in bambu studio

for example, windows use microsoft's edge
linux uses webkit

for your case, it is very likely related with the webkit not workable on bambu studio
are your linux system using webkit?

could you help to upload the log for us to have a check

bambu123, 2023-10-16:
@Tomkee There are some informations here #1828
Could you help check.

SnickCH, 2023-10-16:
I also tested on Garuda with yay to install the bambustudio-bin (from chaotic-aur). No success. 
I think it has to do wit the web-content (how it is presented / loaded) for the login. Unfortunately I couldn't find any information about this

lanewei120, 2023-10-16:
> I also tested on Garuda with "yay" to install the bambustudio-bin (from chaotic-aur). No success. With all the sources, app-images etc. I alsways get the same result.
> 
> I think it has to do wit the web-content (how it is presented / loaded) for the login. Unfortunately I couldn't find any information about this

@SnickCH 

Could you also check the issue https://github.com/bambulab/BambuStudio/issues/1828
and check whether that solution can work or not for you?

SnickCH, 2023-10-19:
@lanewei120 

"WEBKIT_DISABLE_COMPOSITING_MODE=1 ./Bambu_Studio_Linux_Fedora_v01.07.07.89-20231013104019.AppImag"
->I only tested this

With this it starts and I see now the content. But I can't login (kin of there is no connection or it is just not doing anything). On the console I get a lot of warnings and "failed" messages. The output from the console is in the .txt file. 

[bambustudio_logs.txt](https://github.com/bambulab/BambuStudio/files/13038293/bambustudio_logs.txt)

bambu123, 2023-10-19:
> @lanewei120
> 
> "WEBKIT_DISABLE_COMPOSITING_MODE=1 ./Bambu_Studio_Linux_Fedora_v01.07.07.89-20231013104019.AppImage" ->I only tested this
> 
> With this it starts and I see now the content. But I can't login (kind of there is no connection or it is just not doing anything) - when I press the green sign in button it shows the "circle" that's spinning for 1-2 seconds and then nothing happens, not even an error message. On the console I get a lot of warnings and "failed" messages. The output from the console is in the .txt file.
> 
> [bambustudio_logs.txt](https://github.com/bambulab/BambuStudio/files/13038293/bambustudio_logs.txt)
> 
> Edit:
> 
> If I also test this part, the application doesn't start (of course I adjusted the command and path to my version numbers)
> 
> ~ λ ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage --appimage-portable-home ~ λ mkdir Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage.home/.confg ~ λ mv ~/.config/BambuStudio Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage.home/.confg

Have you installed the network plugin?  Login page is empty without network  plugin.

SnickCH, 2023-10-19:
@lanewei120 Do I have to install a network plugin with an app-image (shouldn't it be in the app-image)? Frankly: I tested so much I don't  remember that I was asked to install the network plugin.

I see in the directory ~/.config/BambuStudio a plugin folder with the plugin. But I'm not sure if this was from the app image or from the chaotic-AUR installation. 

The documentation to manually install the networkplugin only shows the information for windows and mac

SnickCH, 2023-10-19:
@bambu123  I deleted everything and restarted from scratch. Now I installed the networkplugin (during the setup). Same problem: I can't login and no error message

SnickCH, 2023-10-19:
To summarize for others with the same (blank wizard) error what I have done to make it run. This was done on Garuda Linux.

- delete any existing config folder: ~/.config/BambuStudio
- Download the newest App-image
- start it with the following parameter: WEBKIT_DISABLE_COMPOSITING_MODE=1 ./Bambu_Studio_Linux_Fedora_v0x.x.x.x.x........AppImage
- Login
- If nothing happens: press "forgot" for the password reset and wait until the wizard for the password reset is loaded
- cancel the password reset wizard (at this point, I was logged in)

closed 2023-10-31 by bambu123

aslwerner, 2024-03-19:
If you have installed it with the flatpak and have the error with the "white start screen", then the whole thing can be easily fixed. 

Open a Terminal :

Normally, the flatpak path is located under -> `var/lib/flatpak`
Here, simply go deeper into the following directory for Bambu Studio. 

`cd /var/lib/flatpak/app/com.bambulab.BambuStudio/x86_64/stable/active/ `

And open the metadata files there with an editor. 

`sudo nano metadata`

then under **Environment** -> enter `WEBKIT_DISABLE_COMPOSITING_MODE=1`

If you like, you can also activate the DARK-MODE here 

```
[Environment]
BAMBU_STUDIO_DARK_THEME=true
WEBKIT_DISABLE_COMPOSITING_MODE=1
```

cdoublejj, 2025-01-11:
Same here windows 11 B studio 1.10.1.50 i know for a fact this version of win11 does not have and i can't seem to install it. alt+f44 got me past setup wizard in to bstudio app

cdoublejj, 2025-01-11:
well the printing profile ios forever stuck as a carbon x1c even though it is connected to an a1 mini, it will not print. i think it defaults to edge as a the rendering engine, this is so fakd, id sooner have linux on here if cricuts weren't garbage based o nwindows sodftware.