bambulab/BambuStudio #2568: no printers and profiles after updating to v1.7.6.91

3DAixperts, 2023-09-22:
### Bambu Studio Version

1.7.6.91

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 22H2 Build 22621,2359

### Printer

X1C

### How to reproduce

Install the actual update

### Actual results

no printers and profiles are shown
same result when settings folder in AppData is removed

### Expected results

should show my printers and filament profiles

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12700123/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

xdigitalfoto, 2023-09-22:
hello, I have the same thing, I only have the default printer and I can't see X1C, nor my profiles for filaments, which I had tried and set to my liking. During one PS resrart, it was possible to display the X1C after the next start without a restart, the default printer again. Every time Bambu Studio starts, it wants to start the setup as if it had never been started. It means that I don't print... the default printer certainly doesn't know the XC1 Gcode settings for start

eXenInfo, 2023-09-22:
Same Problem with me.
Windows 10 
Bambu P1P with AMS
Uninstall this version and install the 1.7.4.52 the issue is not present.

Pecetero, 2023-09-22:
El mismo problema.
He tenido que volver a instalar la versión anterior

mlp3d, 2023-09-22:
Same error after update 1.7.6.91. No profiles loaded. Even after new login and klick on load Profiles from BambuStudio.

BambuFri, 2023-09-22:
Bei mir auch der Fehler nach Update 1.7.6.91.
Hoffe, das wird bald gelöst...

christianpfitzner, 2023-09-22:
Same issue with Windows 11: Each time I boot up, Bambu Studio asks for complete setup; afterward, my X1C is not recognized.

xXemix, 2023-09-22:
Same here in Windows 11, not show profiles and my X1C not recognized, imposible sync with them.

davidaberg, 2023-09-22:
Same for P1P on MacOS

Lerk90, 2023-09-22:
I think the problem is the language, try English.

BambuFri, 2023-09-22:
Ich habe die Sprache geändert und dann neu gestartet und nochmals die Sprache auf meine richtige eingestellt. Keine Ahnung warum, aber es funktioniert... :-)

I changed the language and then restarted and set the language again to my correct one. I don't know why, but it works... :-)

Lerk90, 2023-09-22:
> Ich habe die Sprache geändert und dann neu gestartet und nochmals die Sprache auf meine richtige eingestellt. Keine Ahnung warum, aber es funktioniert... :-)
> 
> I changed the language and then restarted and set the language again to my correct one. I don't know why, but it works... :-)

but if you restart again with your language, fail again.

BambuFri, 2023-09-22:
Yes, I just noticed that too :-(

nargos, 2023-09-22:
gleiches Problem!

12ackcasa, 2023-09-22:
Hello, effectivement un changement de langue en anglais et ensuite la langue de base, ça marche !

christianpfitzner, 2023-09-22:
> Same issue with Windows 11: Each time I boot up, Bambu Studio asks for complete setup; afterward, my X1C is not recognized.

My issue is also solved by changing from German to English.

xdigitalfoto, 2023-09-22:
the official site no longer offers the new version, but I am exhibiting the Bambu_Studio_win-v01.07.04.52.exe version, error will be bigger than it seems

lanewei120, 2023-09-23:
we have solved this issue on v1.7.6.92
please have a try

closed 2023-09-23 by lanewei120