bambulab/BambuStudio #2569: With update to 1.7.6.91 every time welcome screen and no calibration possible

Obicom, 2023-09-22:
### Bambu Studio Version

v1.7.6.91

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Install new version 1.7.6.91
2. Start Bambu Studio
3. Goto Calibration

### Actual results

- After installing the new release 1.7.6.91 it starts with a welcome screen and base setup. This welcome screen appears every time when I start the new version. 
- Flow calibration shows all filaments as "incompatible". 
- Print plate preview (print plate) is to bright in dark GUI mode. Not really good visible print plate

### Expected results

1. No welcome screen and use the already existing settings
2. Flow calibartion should show the right filaments (not "incompatible)
3. Print plate should have more contrast in dark GUI mode

### Project file & Debug log uploads

no log or project file is needed to reproduce the problem. Did the installation on two different computers with the same outcome.
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ImYourFather187, 2023-09-22:
same problem

lanewei120, 2023-09-23:
hello @Obicom 
it seems a problem caused by config file loading fail under some language
we have released v1.7.6.92 to solve this issue,

could you have a try for v1.7.6.92?

Obicom, 2023-09-23:
> hello @Obicom it seems a problem caused by config file loading fail under some language we have released v1.7.6.92 to solve this issue,
> 
> could you have a try for v1.7.6.92?

1.7.6.92 solved my issue.

closed 2023-09-23 by lanewei120