bambulab/BambuStudio #2570: Set up keep showing when oppening the app / hight printing time

LionesC, 2023-09-22:
### Bambu Studio Version

1,7,6,91

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I5 11 gen
16 ram
250 ssd

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Open the app -> set up shows

Printing time:
Open new project -> add stl file -> print setings as another file -> print time 5h
If using the printing file by deleting print and adding the same stl file -> print time 1h

Printer and filament tipe doesn't show in setings. Only "Default printing" and "Default filament"

### Actual results

Printing time is x5
Can't select filament or printer

### Expected results

as mentioned up

### Project file & Debug log uploads

[Bambu.zip](https://github.com/bambulab/BambuStudio/files/12702643/Bambu.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-09-23:
should be the same issue as https://github.com/bambulab/BambuStudio/issues/2568

please try v1.7.6.92

lanewei120, 2023-09-27:
should be clodes in v1.7.6.92

closed 2023-09-27 by lanewei120