bambulab/BambuStudio #2572: Bambu Studio Calibration does not see the AMS in the english version

toml899, 2023-09-22:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1 Carbon

### How to reproduce

Calibration > Flow Dynamics OR Flow Rate > Manual OR Auto-Calibration

AMS filaments are not detected

### Actual results

AMS filaments are not detected

### Expected results

AMS filaments should be detected

### Project file & Debug log uploads

[debug_Fri_Sep_22_12_29_54_3408.log.0.zip](https://github.com/bambulab/BambuStudio/files/12702940/debug_Fri_Sep_22_12_29_54_3408.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-09-25:
@toml899  Could you use a picture to express your problem?

toml899, 2023-09-25:
Hi, here are two pictures.  1 - is what the AMS has.  2 - Is what Bambu studio has.

Please help me fix.  Grand children are most upset!
[image]
[image]

toml899, 2023-09-25:
This worked fine until about a week ago.

zhimin-zeng-bambulab, 2023-09-26:
@toml899 
From the picture, you are using external material to print. When you use external material, only the external material will be displayed, and AMS will not be displayed. In addition, you need to set the material type of the external material on the device page before you can select and set it on the calibration page.

zhimin-zeng-bambulab, 2023-09-26:
If you want to use AMS materials for calibration, you can first unload the external material and then use AMS materials for calibration.

toml899, 2023-09-26:
Thanks.  That fixed the issue

closed 2023-09-27 by zhimin-zeng-bambulab