bambulab/BambuStudio #2573: BambuStudio crashes before splash screen - unusable

tennr, 2023-09-22:
### Bambu Studio Version

1.7.6

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.5

### Printer

X1C

### How to reproduce

attempt to open BambuStudio... observe the crash report dialog

### Actual results

crash

### Expected results

application opens

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-09-23:
@tennr 

are you using v1.7.6.92?
and does it crash when the splash screen is shown?

could you help to fetch the crash stack following this wiki https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue?

tennr, 2023-09-29:
```json
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [84963]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.07.06.92 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-09-22 13:41:04.8381 -0700
OS Version:            macOS 13.5 (22G74)
Report Version:        12
Anonymous UUID:        BEFBDA34-E1D6-3BE2-C81D-6A698AF9CA6F

Sleep/Wake UUID:       CA727BD6-772F-4D9F-A3B2-A3CEA327D8EE

Time Awake Since Boot: 4000000 seconds
Time Since Wake:       2843836 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000005a0
Exception Codes:       0x0000000000000001, 0x00000000000005a0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [84963]

VM Region Info: 0x5a0 is not in any region.  Bytes before following region: 105553518918240
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x101b7802c Slic3r::GUI::MainFrame::update_side_preset_ui() + 60
1   BambuStudio                   	       0x101b3490c Slic3r::GUI::GUI_App::reload_settings() + 132
2   BambuStudio                   	       0x101b5caa8 wxAsyncMethodCallEventFunctor<Slic3r::GUI::GUI_App::start_sync_user_preset(bool)::$_49::operator()(bool) const::'lambda'()>::Execute() + 244
3   BambuStudio                   	       0x103b45b30 wxEvtHandler::ProcessEventLocally(wxEvent&) + 172
4   BambuStudio                   	       0x103b45900 wxEvtHandler::ProcessEvent(wxEvent&) + 56
5   BambuStudio                   	       0x103b45750 wxEvtHandler::ProcessPendingEvents() + 236
6   BambuStudio                   	       0x103aa997c wxAppConsoleBase::ProcessPendingEvents() + 112
7   BambuStudio                   	       0x103ace5c8 wxEventLoopBase::DoYieldFor(long) + 52
8   BambuStudio                   	       0x103ace51c wxEventLoopBase::YieldFor(long) + 120
9   BambuStudio                   	       0x103aa9350 wxAppConsoleBase::Yield(bool) + 104
10  BambuStudio                   	       0x103864d14 wxWebViewWebKit::RunScriptSync(wxString const&, wxString*) const + 248
11  BambuStudio                   	       0x103865148 wxWebViewWebKit::RunScript(wxString const&, wxString*) const + 84
12  BambuStudio                   	       0x103861c78 wxWebView::GetUserAgent() const + 76
13  BambuStudio                   	       0x101dbd5a0 Slic3r::GUI::PrinterWebView::PrinterWebView(wxWindow*) + 2200
14  BambuStudio                   	       0x101b64660 Slic3r::GUI::MainFrame::init_tabpanel() + 1204
15  BambuStudio                   	       0x101b62db4 Slic3r::GUI::MainFrame::MainFrame() + 2868
16  BambuStudio                   	       0x101b22e88 Slic3r::GUI::GUI_App::on_init_inner() + 5704
17  BambuStudio                   	       0x101b217f4 Slic3r::GUI::GUI_App::OnInit() + 16
18  BambuStudio                   	       0x103a03eb8 wxApp::CallOnInit() + 152
19  BambuStudio                   	       0x103afbe48 wxEntry(int&, wchar_t**) + 128
20  BambuStudio                   	       0x101b0c010 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 172
21  BambuStudio                   	       0x100fc511c Slic3r::CLI::run(int, char**) + 23128
22  BambuStudio                   	       0x100ffe22c main + 172
23  dyld                          	       0x19395bf28 start + 2236

Thread 1:
0   libsystem_pthread.dylib       	       0x193caed8c start_wqthread + 0

Thread 2:
[... 458 lines cut ...]
    },
    {
      "rolloutId" : "62699e1ec1ff2978b47f6c3b",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "631f72d1de559130376df1fb"
      },
      "deploymentId" : 240000023
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "3a3cf641-8471-4e4e-9ad4-81d0ede970fd",
      "experimentId" : "64a84dae90d82611a0bd7d3d",
      "deploymentId" : 400000010
    },
    {
      "treatmentId" : "6dd670af-0633-45e4-ae5f-122ae4df02be",
      "experimentId" : "64406ba83deb637ac8a04419",
      "deploymentId" : 900000017
    }
  ]
}
}

```

lanewei120, 2023-09-29:
@tennr thanks very much for this detailed stack
we will have a more debug on this stack

is this issue 100% reproduciable on your side?
if yes, you can remove /Users/user.name/Library/Application Support/BambuStudio/plugins and reboot bambustudio

tennr, 2023-09-29:
@lanewei120 It failed at least 6 times in a row. I had to uninstall and reinstall previous version and restore Studio settings from a backup.

Do you know if removing plugins is a known fix for this behavior?

lanewei120, 2023-10-03:
> @lanewei120 It failed at least 6 times in a row. I had to uninstall and reinstall previous version and restore Studio settings from a backup.
> 
> Do you know if removing plugins is a known fix for this behavior?

@tennr 
currently this is the first one found with this callstack
as we can not reproduce it here, 
I am afraid we don't know whether it can fix it or not
just for a suggestion to try

bambu123, 2023-10-07:
@StoneLiBambulab 可能调用 wxWebViewWebKit::RunScript 太早了。GetUserAgent 相关调用需要延后。

StoneLiBambulab, 2023-10-08:
it will be fixed in next release.

Thanks!

Stone Li

SaltWei, 2024-01-18:
@tennr Does this issue still exist under your PC environment?