bambulab/BambuStudio #2574: Device tab video width not scaling with window

LJKraus, 2023-09-22:
### Bambu Studio Version

1.7.6.92  

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

WIndows 11 Pro 22H2  build 22621.2283

### Additional system information

32Gb memory, Nvidia GTX1070 8Gb video memory, 4k screen, normal 100% scaling

I found the same behavior using a different PC: Windows 10 Pro, 4Gb RAM, 1920x1080 screen, 100% scaling

### Printer

Bambu Lab X1C

### How to reproduce

Open Studio in a window, switch to Device tab, video monitor area appears with printer controls on the right.
[image]

Resize the program window in any amount, in any direction, larger or smaller.

### Actual results

The video monitor area expands to ~1610 pixels in width, and cannot be narrowed.   No further resizing of the program window will reduce the new width of the video monitor area. 

The only way I've found to restore a properly proportioned video monitor is to restart Studio.  

If the program window is resized _before_ opening the Device tab, the width of the video monitor will adjust as expected.

If Studio was full screen when last shut down, and therefore opens full screen, it can be windowed and resizing the window from the Device tab works as expected.  If Studio  opens windowed and is then resized while on the Device tab, the error occurs.

The issue existed in previous versions of Studio, but I had not taken the time to rule out possible causes in my system (scaling, 4K resolution).  

### Expected results

Video monitor area should resize proportionally with the program window.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12704743/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LJKraus, 2023-09-22:
Forgot to paste "Actual" example:
[image]

bam-winterfell, 2023-09-25:
Hi.@LJKraus
This issue will be resolved when the next version is released. Please update it in a timely manner and your issue will be easily resolved.Before that, you can try to use a shorter file name.
Wishing you a happy life

bam-winterfell, 2023-11-14:
@LJKraus The new version has been released on the release page and has fixed this issue. You can have a try.

closed 2023-11-14 by bam-winterfell