bambulab/BambuStudio #2575: Stop on layer height time estimation

dilwei3374, 2023-09-22:
**Is your feature request related to a problem? Please describe.**

It would be nice if Bambu Studio would give a time estimation on when a "pause" would occur. For example, recently I made a project where I needed to insert a magnet into my print. However, Bambu Studio would not give me a time estimation on when this pause would occur. Therefore I couldn't set a timer and had to wait near the printer for the pause to occur.

PrusaSlicer DOES have this feature.

**Describe the solution you'd like**

After selecting "Slice", the information menu would show a "time to pause" time next to or near the "overall print time". It would be nice if this was displayed on the printer as well

**Describe alternatives you've considered**
NA

**Additional context**
NA

LJKraus, 2023-09-22:
Estimated time to layer would be nice to have.

Meanwhile, periodically checking the print progress with the Handy app lets me do other things while waiting for the correct layer.

dilwei3374, 2023-09-24:
> Estimated time to layer would be nice to have.
> 
> Meanwhile, periodically checking the print progress with the Handy app lets me do other things while waiting for the correct layer.

That's true. I am just used to this feature in PrusaSlicer and found it odd that it's not a feature in BambuStudio as its based of PrusaSlicer. Hopefully it can be added at somepoint.

RaphTronic, 2023-12-15:
Estimated time to Pause would indeed be useful. I often use Layer Pause to investigate print issues. Especially when trying to dial in Flow & Overlap with new filaments.

Unfortunately, with Bambu Studio I have to set a timer on my computer to ballpark when to check on the printer. If set too short I have to wait for it to get to the pause point, and it slows down the development time. If set too long, time was lost as the printer was paused for a while, and that slows down the filament tuning process.

Quite cumbersome and inefficient. Would rather have an estimate as to how long before the Pause Layer will likely kick in.

TaylorN15, 2024-01-04:
What about a notification when the pause happens? We already get notifications when manually paused, why not for a layer pause too?

mpetito, 2024-01-28:
Related to this, it would be nice if the mobile app could tell you how much time is remaining until the next gcode pause will occur for the current print.

punkman, 2024-06-19:
I was looking for a solution to estimate time to pause and I found that skimming on the right layer slider it shows a print timer.
I didn't checked the precision with an actual print but indeed could be useful to estimate when to go and check the printer.

This doesn't solve the missing time estimation for layer end/pause in Handy app but a timer could be enough.
[image]