bambulab/BambuStudio #2576: Purge Block Material Estimation

dilwei3374, 2023-09-22:
**Is your feature request related to a problem? Please describe.**

During a multi-material print, Bambu Studio will give a material estimation on the amount of "flush" being consumed. However, there is no such estimation on the amount of material being used when using a purge tower. Would be nice to know how much material I'm wasting.

**Describe the solution you'd like**

In the estimation window, there should be a "purge tower" section next to "flush" that estimates the amount of material being used in just the tower.

**Describe alternatives you've considered**
NA

**Additional context**
NA

MadMax389, 2023-09-24:
It's actually a prime tower, not a purge tower. In any case, it is calculated as part of the model weight. The only workaround that I know of is to slice it with and without the prime tower and note the difference in model weight.

dilwei3374, 2023-09-24:
> It's actually a prime tower, not a purge tower. In any case, it is calculated as part of the model weight. The only workaround that I know of is to slice it with and without the prime tower and note the difference in model weight.

That's fair, however I wish that I wouldn't need to do that in order to calculate the difference. Will keep this in mind in the future though.

XunZhangBambu, 2023-11-14:
In 1.8 beta , wipe tower filament consumption has been seperated from model .
[image]

closed 2023-12-21 by QingZhangBambu