bambulab/BambuStudio #2577: More 0.2 and 0.8mm nozzle presets

dilwei3374, 2023-09-22:
**Is your feature request related to a problem? Please describe.**
Would be nice if the 0.2mm and 0.8mm nozzles had more presets like the 0.4mm nozzle

**Describe the solution you'd like**
Have a similar preset system like the 0.4mm nozzle for the 0.2mm and 0.8mm nozzles 

**Describe alternatives you've considered**
NA

**Additional context**
NA

QingZhangBambu, 2023-09-25:
Do you mean P Series? We are testing it, which will be added as all data is collected.

dilwei3374, 2023-09-26:
> Do you mean P Series? We are testing it, which will be added as all data is collected.

I do mean P Series, specifically P1P. If it helps, I've been using 0.03mm layer height on the 0.2mm nozzle just fine so far. Appreciate the reply

QingZhangBambu, 2023-10-09:
we've tested it, and it will be added in next official version