bambulab/BambuStudio #2578: A1 Mini slice has travel moves to the right of the bed even when using a single filament

kvnper, 2023-09-22:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

windows10

### Additional system information

intel cpu

### Printer

a1 mini 0.4n

### How to reproduce

slice attached project file

### Actual results

travel moves to the right of the printer bed every layer when using a single filament

[image]

### Expected results

no travel moves to the right of the bed

### Project file & Debug log uploads

[a1 mini bug.zip](https://github.com/bambulab/BambuStudio/files/12704943/a1.mini.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-09-23:
This is not bug, you can disable timelapse when send printing job, and the printer will not go to the side of bed to take photo

kvnper, 2023-09-23:
> This is not bug, you can disable timelapse when send printing job, and the printer will not go to the side of bed to take photo

I understand. These travel moves are being added to the print time, perhaps it should be like the X1C behaviour.

SaltWei, 2023-09-23:
It's difficult.
A1 mini is i3 structure, and the bed is always moving during printing, which cause the traditional timelapse very ugly.
So specific timelapse gcode is add for i3 structure.

X1C is coreXY and the bed is always fixed

You can directly delete the timelapse gcode for A1 mini to delete the travel, but attention, it will not take photo anymore
[image]

SaltWei, 2023-09-23:
By the way, this will not cause much time. For 90 layers, the total time only increase less than 2 minutes. Not important

pKoeni, 2023-12-21:
I also stumbled across this, the problem is not just the travel time, also its creating stringing and sometimes the printer decides to do an automatic nozzle cleaning, which leads to extra poop and a worse print result. It would be much easier if there was a checkbox to activate/deactivate the time lapse instead of just selecting the modes and manually deleting the G code.
[image]

SaltWei, 2023-12-22:
> It would be much easier if there was a checkbox to activate/deactivate the time lapse

User can disable timelapse when send printing job
[image]

mike-odom, 2024-01-14:
Issue marked as bug for the timelapse feature travel times in https://github.com/bambulab/BambuStudio/issues/3161

TylonHH, 2024-02-26:
> User can disable timelapse when send printing job
But when I slice the object I already see the moves. After sendig THIS I can also uncheck the timelapse. I never use timelapse.
[image]

TylonHH, 2024-02-26:
### Workaround: Print sequence -> by object
this disable the timelapse

Zanderama, 2024-07-31:
This totally is a 'bug' or at the very least a very poor 'feature'.  It seems the slicer for A1/mini adds this mess of excessive travel, then unticking 'timelapse' filters this out from the g-code so it doesn't actually happen.  However, this 'bug' stops you accurately seeing in your slicer what difference changes make to real travel, speed, stringing etc.  Instead the obvious things are to 1) Add an option for Timelapse 'Off' to "Traditional' and 'Smooth'.  or 2) Only actually add the timelapse g-code when the timelapse check-box is 'on'.

Printing 'by object' is a hacky workaround (thanks) but doesn't help printing a batch of items together - unless you then perhaps hackily merge them together as one object then print that as a 'by object'.  All very hacky for what's an obvious bug, and not obvious to a new user to Bambu Slicer.

TylonHH, 2024-09-15:
Or use print per object, this also disable the Timelapse.

mike-odom, 2024-09-16:
+1 for this being a bug. The calculations for how long a print are grossly exaggerated by this timelapse feature. I'd like to know how long my print will take.

Edya7659, 2024-09-25:
Couldn't you just add "Disable" to the Special Mode/Timelapse menu, and then the movement of the print head to the shooting position would not be calculated (disappeared)? Yes, in that case, there would be NO possibility of enabling Timelapse shooting when sending the model to print. But, on the other hand, it would be possible to get more accurate data on the model's printing time.

[image]

Kaskadeur111, 2025-01-08:
As a new A1 user comming from prusa... i want to disable this gcode BEFORE sending to the printer ... too.

f2l2pe, 2025-05-13:
I have the same problem, everything was disabled and the printer still parked after every layer. The only way to really disable this is removing the timelapse gcode.