bambulab/BambuStudio #2579: Bambulab Studio Crashes when trying to open the camera with a dedicated GPU

XhanceHd, 2023-09-23:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

Bambu Lab P1S

### How to reproduce

Setting the bambulab studio to run on dedicated GPU on a Laptop

### Actual results

Client Crash

### Expected results

NA

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ffafe5, 2023-09-23:
I am also seeing this bug after the latest Maker World update. I was able to view the camera yesterday after initially updating, but this morning opening the Device tab crashes Bambu Studio.

ffafe5, 2023-09-23:
I did find that moving Bambu Studio to a different monitor (also on the same GPU video outputs, not motherboard) fixed the issue. I have 4 monitors. The original crashing monitor is a side monitor, and the non-crashing monitor is my primary display set in Windows. Both monitors are the same resolution, though the other 2 monitors in my system are unique resolutions.

bambu123, 2023-09-25:
It may still be a problem with the D3D9 library. Please refer to the wiki below to send us the DUMP file.

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

kluke3, 2023-09-25:
I'm a different user with the same issue. Program crashes when trying to open the camera with a dedicated GPU.
I've attached the zipped up .dmp file. Git hub wouldn't let me upload it directly, so I zipped it  first.
[bambu-studio.exe.4872.dmp.zip](https://github.com/bambulab/BambuStudio/files/12715972/bambu-studio.exe.4872.dmp.zip)

bambu123, 2023-09-26:
> I'm a different user with the same issue. Program crashes when trying to open the camera with a dedicated GPU. I've attached the zipped up .dmp file. Git hub wouldn't let me upload it directly, so I [...]

Same As  #1361, #909 .
Please refer https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

hewfish, 2023-09-28:
I also have the same issue, when using RDP to control my machine that has BambuStudio. 1.7.4.52 works fine. Upgraded to latest 1.7.6 and crashes as soon as print job is sent and it switches to device view. Attached a bunch of dump files from different tests.
[bambu-studio.exe(1).183928.dmp.zip](https://github.com/bambulab/BambuStudio/files/12745296/bambu-studio.exe.1.183928.dmp.zip)
[bambu-studio.exe(1).185944.dmp.zip](https://github.com/bambulab/BambuStudio/files/12745298/bambu-studio.exe.1.185944.dmp.zip)
[bambu-studio.exe(1).215108.dmp.zip](https://github.com/bambulab/BambuStudio/files/12745301/bambu-studio.exe.1.215108.dmp.zip)
[bambu-studio.exe.170236.dmp.zip](https://github.com/bambulab/BambuStudio/files/12745302/bambu-studio.exe.170236.dmp.zip)

bambu123, 2023-09-28:
> I also have the same issue, when using RDP to control my machine that has BambuStudio. 1.7.4.52 works fine. Upgraded to latest 1.7.6 and crashes as soon as print job is sent and it switches to [...]

These DMP files show another issue that has nothing to do with liveview. Do you want to create a new issue?

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works