bambulab/BambuStudio #2582: New version of Bambu studio generates a M600 causing my kobra max freeze

lpp12138, 2023-09-23:
### Bambu Studio Version

01.07.06.92

### Where is the application from?

Bambu Lab github releases

### OS version

windows10

### Printer

Anycubic Kobra Max

### How to reproduce

Updated from BambuStudio 01.07.00.65 to BambuStudio 01.07.06.92
In new version it generates Gcode with a M600 which freeze the printer after running it.
Delete it will make printer work as excpeted. 
External filament sensor may causing command M600 never ends , because filament will always inside sensor when printing starts with such broden extruder. 

### Actual results

Printer got freezed

### Expected results

Printer work as excepted

### Project file & Debug log uploads

[g.zip](https://github.com/bambulab/BambuStudio/files/12706009/g.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-09-25:
@lpp12138 is kobra Max doesn't support M600? how about M601?

QingZhangBambu, 2023-09-25:
i compared gcode between each version. Two version has M600 both.

QingZhangBambu, 2023-09-25:
[image]

QingZhangBambu, 2023-09-25:
Could the version's gcode work normally?

lpp12138, 2023-09-25:
box-restart.gcode is generate by 01.07.06.92 and box-改6-1.gcode is generate by 01.07.00.65. It's wierd that in your tests both version do the same thing. I will test for M600 and M601,wait for 30minutes. :)

lpp12138, 2023-09-25:
[image]
For the original box-restart.gcode. M600 causing the exruder draw back some filaments, play a beepbeep music, then stucks.
Press pause button after music will make the screen stucks at "waiting last command ends".
Delete M600 or change it to M601 will make printing starts normally, no beepbeep music.

QingZhangBambu, 2023-09-26:
anyway, i will change M600 to M601, so that kobra will not stucks.

closed 2023-10-24 by QingZhangBambu